dallmann-consulting/OCPP.Core #1: Build the envorinment

tomas890628, 2021-02-05:
Hi,First thanks for the great work.
I really like this UI and some of the things.
I'm a student from taiwan and now trying to build a charging station.
Here is my problem :
First I follow the step in Configuration and Installation:
STEP 1 & 2 I use the default , I didn't do any changes.
STEP 3:
Make sure you installed the .NET-Core SDK.           ###  I download : .NET Core 3.1 SDK (v3.1.405)
Open a command shell and navigate to the root folder of the project.   ### Yes
Execute the following command dotnet build OCPP.Core.sln    ### Success
Copy the 'wwwroot' folder to the build target directory (where the dlls are)   
### I can't understand this,so I copy the folder to 
- OCPP.Core.Database\bin\Debug\netcoreapp3.1
- OCPP.Core.Management\bin\Debug\netcoreapp3.1
- OCPP.Core.Server\bin\Debug\netcoreapp3.1

Start both executables (=> Kestrel webserver)   ### This step I  can't understand either.

I have learn C# windowform before , I want to see this project's feature can do work for me or not.
Thank you!

dallmann-consulting, 2021-02-05:
Hi,

 

do you need to work with the SDK?

Maybe its simpler to open the project in Visual Studio (Community Edition) and just hit “run”. But of course this is for testing purposes mainly.

 

If you want to run it on a server or something it would be best to install it on IIS:

https://codingsonata.medium.com/deploy-asp-net-core-web-api-on-iis-f75e755a6402

https://stackify.com/how-to-deploy-asp-net-core-to-iis/

 

The compiled output of the two web projects contains “.exe” files, right?

This way you’re starting the simple webserver “kestrel”:

https://docs.microsoft.com/de-de/aspnet/core/fundamentals/servers/?view=aspnetcore-5.0 <https://docs.microsoft.com/de-de/aspnet/core/fundamentals/servers/?view=aspnetcore-5.0&tabs=windows> &tabs=windows

 

You can simply start the exe files und will get an output like this:

info: Microsoft.AspNetCore.DataProtection.KeyManagement.XmlKeyManager[0]

      User profile is available. Using '…path…\DataProtection-Keys' as key repository and Windows DPAPI to encrypt keys at rest.

info: Microsoft.Hosting.Lifetime[0]

      Now listening on: http://localhost:5000

info: Microsoft.Hosting.Lifetime[0]

      Now listening on: https://localhost:5001

info: Microsoft.Hosting.Lifetime[0]

      Application started. Press Ctrl+C to shut down.

 

For the management UI just open http://localhost:5000 in a browser and you should see the login page.

Very easy.

 

But the kestrel server is not intended for internet access. You should use a reverse proxy before that. That’s why it may be better to directly install the application in IIS. You have much more control with accounts, SSL-certificates and settings there.

 

I will try to write a more detailed installation instruction.

tomas890628, 2021-02-05:
Hi,Thanks for your reply
I'll try it in next monday,If there have any question or anything update.
I'll update here
Thanks!

dallmann-consulting, 2021-02-06:
See new Build & Run instructions:
https://github.com/dallmann-consulting/OCPP.Core/blob/main/Installation.md

tomas890628, 2021-02-09:
Hi,sorry for the late reply.
I have follow the  new Build & Run instructions
And on the first step:
[Database](https://github.com/dallmann-consulting/OCPP.Core#configuration-and-installation)
I can't really understand this(I usually use xampp to do the local work, is this work for me?) 
so I didn't do anything,then I go ahead to the next:
[Build and run](https://github.com/dallmann-consulting/OCPP.Core#build--run)
this step is work for me , and I can go to the login page, and I login by the account:admin/t3st
But when I login , I get the error message like below:
------------------------ERROR CODE----------------------------------
Error
An error occurred while processing your request.
Request ID: |e866a0a3-45e630a15d19bce7.

Development Mode
Swapping to Development environment will display more detailed information about the error that occurred.

The Development environment shouldn't be enabled for deployed applications. It can result in displaying sensitive information from exceptions to end users. For local debugging, enable the Development environment by setting the ASPNETCORE_ENVIRONMENT environment variable to Development and restarting the app.
------------------------------------------------------------------------
which step  did I do wrong?
Thanks for your reply!

dallmann-consulting, 2021-02-09:
Hi,

 

you’re almost there 😊

 

That is a problem with the database storage.

 

Both applications (OCPP-Server and Management UI) store their data in the same database. You can use a SQL-Server (SQL-Express is free) or the simpler SQLite database. SQLite is just a simple file and both applications need to access the same file.

 

The “appsettings.json” files in both applications contain the following section:

 

  "ConnectionStrings": {

    //"SQLite": "Filename=.\\..\\SQLite\\OCPP.Core.sqlite;"

    "SqlServer": "Server=.;Database=OCPP.Core;Trusted_Connection=True;"

  },

 

Only one connection can be active. In the above case the “SQLite” is inactive. If you want to use SQLite change it to something like this in BOTH applications:

 

  "ConnectionStrings": {

    "SQLite": "Filename=C:\\temp\\OCPP.Core.sqlite;"

  },

 

Then copy the file “OCPP.Core\SQLite\ OCPP.Core.Empty.sqlite” to the specified location and remove the “empty” in the filename.

 

Now it should work!

dallmann-consulting, 2021-02-11:
Does it work or do you need further help?

tomas890628, 2021-02-19:
Sorry for the late reply, last 2 week is chinese new year,
I'll follow your new build&run if I found new questions, I will let you know
In addition,is this system work with the real charging machine? I have 2~4 charging machine(use OCPP) here.

tomas890628, 2021-02-19:
Hi I have try the new build & run
First I install the sqlite , I use this tutorials
https://www.runoob.com/sqlite/sqlite-installation.html
Then change the code in mangerment and server application.json:

  "ConnectionStrings": {
    //"SQLite": "Filename=.\\..\\SQLite\\OCPP.Core.sqlite;"
    //"SqlServer": "Server=.;Database=OCPP.Core;Trusted_Connection=True;"

    "SQLite": "Filename=C:\\temp\\OCPP.Core.sqlite;"

  },
Create the folder C:/temp

Copy the file in \OCPP.Core\SQLite\OCPP.Core.Empty.sqlite

Delete the Empty in filename
So it look like this:
C:\temp\OCPP.Core.sqlite

And I still get the same error , I there anything I did wrong?

Thanks again for your help

dallmann-consulting, 2021-02-19:
Hi

 

First of all: happy new year 😊

 

Then to your other question:  Our multifamily house has ordered 4 charge stations but we are still waiting for them to be installed. There is currently a high demand in Germany because of a state funding. We probably have to wait another few weeks. It would be great if you can test it with real charge points!

 

Your error is strange. When I’m doing the exact same thing it works.

But you are saying that you still get the same error message. I had changed the error page to show some details. The error page was still the original from the visual studio template and obviously it is not very helpful.

So I changed the error page. Can you use the latest version? Hopefully the technical error message will help us find the problem.

 

Good luck!

 

https://www.runoob.com/sqlite/sqlite-installation.html

tomas890628, 2021-02-19:
Hi thanks for your reply,
I have rebuild the file and it works now,
Can I know how can I connect to the EV simulator ( Just for test , can you recommend ?) and the real EV charger?
Now I can get Delta、Phihong、Siemens's EV charger , 
If I can test for you guys it will be great!

dallmann-consulting, 2021-02-19:
Hi,

you’re asking how to configure a charge station to use “our” OCPP-Server, correct?

That depends on the charge station.  I found this manual as an example:

https://www.ifix-solar.info/wp-content/uploads/2017/02/KeContact_P30_eu_config_ml1.pdf

On page 59 the OCPP configuration is described in English. In this case you need to enter a base URL (like http://xxxxxx/something) and a chargepoint identity. OCPP specifies that the ID is appended to the base URL.

(The linked manual is a bit old but the current versions don’ t show any screenshots)

The charge station will then use  “http://xxxxxx/something/identity ” as the server URL. And this is exactly what you enter in the chargepoint simulators, right? The identity is important and you must create a chargepoint with that identity in the OCPP.Core Web-UI.

In other charge stations you need specify this differently and mostly you will need to select a protocol version. The OCPP.Core server supports only 1.6 und 2.0 with the JSON profile. The older versions are often SOAP based but this is deprecated in newer OCPP versions.

Does this help you?

dallmann-consulting, 2021-02-19:
The delta chargers obviously  have no Web-UI. They offer a web application instead to create a config file:
http://evcs.deltaww.com/ConfigTools/

http://support.delta-es.com.au/wp-content/uploads/2020/03/AC-Mini-Plus-Installation-manual-3G-and-WIFI.pdf
Configuration
1. Create configuration files via Web configuration tool.
Features and attributes of the product can be configured are as below:
Communication attributes of 3G, Ethernet, WiFi
**OCPP** Version (1.5 SOAP, **1.6 JSON**)
Maximum charging time
More information and instructions are available at **http://evcs.deltaww.com/ConfigTools/.**
2. Download and store the firmware package on a USB flash drive.
3. Configure the product:
a. Make sure the charging progress is disabled.
b. Insert the USB flash drive to the USB port.
c. All indicators (Power / Status / Charge / Fault) illuminate at the same time signifying the process is enabled.
d. A regular slow blink of Power and Status indicators signify a completed
operation.
e. Unplug the USB flash drive.
f. The product restarts automatically, and then returns to the previous status.

tomas890628, 2021-02-23:
OK ,Thanks you , I'll try my best to do this.
If I get any new problems,I'll update here
If you guys have try the real ev charger you can let me know too!
Thanks again!

tomas890628, 2021-02-23:
Hi,I have read these documents roughly, I don't know my thought is correct or not.
All of our ev charger have been Installation、connected by Internet cable and we can connected it by 4G Router.
Can I just use my computer to connect them , and search them in Local area network?
Just like this:
EV charger-------RJ45 cable-------computer 
Then run your OCPP-Server ,connect data from it?

tomas890628, 2021-02-23:
Delta: https://www.deltaww.com/en-US/products/EV-Charging/3906
Phihong:https://www.phihong.com.tw/index.php?route=product/product&path=60_112&product_id=758
Siemens:https://new.siemens.com/global/en/products/energy/medium-voltage/solutions/emobility/versicharge.html

Here is the link about the ev charger we have,I do my best to find these in English , hope this could help!

closed 2021-04-05 by dallmann-consulting

SHIUBING, 2024-05-14:
@tomas890628  Hello, I am also using a Delta charging station recently. As for creating a configuration file through the web configuration tool, only dealers can set it up, and users themselves cannot set it up?