dallmann-consulting/OCPP.Core #2: Deploy server on Azure

itsenes, 2021-02-19:
Hi and thank you for the sample implementation! It gave me a much better understanding of the protocol(s). I tried to deploy the server part on azure and perform some tests but the web socket connection kept disconnecting with error 1006 and no description of course :( . The web sockets protocol is switched on, keep alive is also on - the server works great on my machine btw - tested both with the simulator and a real cp. Do you have any ideas regarding this issue ? Thank you in advance!!!

dallmann-consulting, 2021-02-19:
Hi,
thanks for your feedback! Can you tell me with what real charge points you have already tested it?

I haven't tried it on Azure so far. This is still on my to-do list :-) I have only tested it on IIS which worked fine.
As I'm planning to host it on my own machines at home (just for myself and our neighbours) Azure had low priority.

The error 1006 means "closed abnormally".  You can find some similar issues with SignalR but I couldn't find any obvious reason. 
I'll have to see if I can spend some time trying to deploy it on Azure.

dallmann-consulting, 2021-02-20:
Hi,
I got it working in Azure. I had some trouble with a missing runtime dll but that seemed to be my mistake.  When the application finally startet it all worked.
 
```
Platform: 32 bit
Pipeline: integrated
Http: 1.1
Web Sockets: On
Always on: On
ARR: On
```

Are you sure the OCPP-Server is really running? Wenn you call the root page ("/") in a browser you should get a message (in the latest versions) like:

```
Running...
0 chargepoints connected
```

Maybe the logging configuration or the message dump (I deactived that) causes problems?

itsenes, 2021-02-21:
hi and thank you so much for your prompt response, yes it was an issue with the message dump dir, with a minor refactoring now it works great! thank you so much. The azure settings you mention are correct but i think we could without the ARR which i will test first chance i get and let you know if you want. Thanks again!

dallmann-consulting, 2021-02-21:
Great!
Do you have (dump) messages from a real charging process?
I would like to display the progress in the chargepoint tiles but I need some sample messages for that.
Thanks for your support

closed 2021-04-05 by dallmann-consulting