dallmann-consulting/OCPP.Core #3: install on ubuntu

vasya00044, 2021-05-19:
Hi. Can I install on ubuntu?
I have installed the SDK. Compiled projects (dotnet build OCPP.Core.sln and dotnet publish OCPP.Core.sln).
But when I run ( dotnet run )  I get the error:

crit: Microsoft.AspNetCore.Server.Kestrel[0]
      Unable to start Kestrel.                                                                                                                                
Interop+Crypto+OpenSslCryptographicException: error:2006D080:BIO routines:BIO_new_file:no such file                                                           
   at Interop.Crypto.CheckValidOpenSslHandle(SafeHandle handle)                                                                                               
   at Internal.Cryptography.Pal.OpenSslX509CertificateReader.FromFile(String fileName, SafePasswordHandle password, X509KeyStorageFlags keyStorageFlags)      
   at System.Security.Cryptography.X509Certificates.X509Certificate..ctor(String fileName, String password, X509KeyStorageFlags keyStorageFlags)              
   at System.Security.Cryptography.X509Certificates.X509Certificate2..ctor(String fileName, String password)                                                  
   at Microsoft.AspNetCore.Server.Kestrel.KestrelConfigurationLoader.LoadCertificate(CertificateConfig certInfo, String endpointName)                         
   at Microsoft.AspNetCore.Server.Kestrel.KestrelConfigurationLoader.Load()                                                                                   
   at Microsoft.AspNetCore.Server.Kestrel.Core.KestrelServer.ValidateOptions()                                                                                
   at Microsoft.AspNetCore.Server.Kestrel.Core.KestrelServer.StartAsync[TContext](IHttpApplication`1 application, CancellationToken cancellationToken)        
Unhandled exception. Interop+Crypto+OpenSslCryptographicException: error:2006D080:BIO routines:BIO_new_file:no such file                                      
   at Interop.Crypto.CheckValidOpenSslHandle(SafeHandle handle)                                                                                               
   at Internal.Cryptography.Pal.OpenSslX509CertificateReader.FromFile(String fileName, SafePasswordHandle password, X509KeyStorageFlags keyStorageFlags)      
   at System.Security.Cryptography.X509Certificates.X509Certificate..ctor(String fileName, String password, X509KeyStorageFlags keyStorageFlags)              
   at System.Security.Cryptography.X509Certificates.X509Certificate2..ctor(String fileName, String password)                                                  
   at Microsoft.AspNetCore.Server.Kestrel.KestrelConfigurationLoader.LoadCertificate(CertificateConfig certInfo, String endpointName)                         
   at Microsoft.AspNetCore.Server.Kestrel.KestrelConfigurationLoader.Load()                                                                                   
   at Microsoft.AspNetCore.Server.Kestrel.Core.KestrelServer.ValidateOptions()                                                                                
   at Microsoft.AspNetCore.Server.Kestrel.Core.KestrelServer.StartAsync[TContext](IHttpApplication`1 application, CancellationToken cancellationToken)        
   at Microsoft.AspNetCore.Hosting.GenericWebHostService.StartAsync(CancellationToken cancellationToken)                                                      
   at Microsoft.Extensions.Hosting.Internal.Host.StartAsync(CancellationToken cancellationToken)                                                              
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)                                            
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)                                            
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.Run(IHost host)                                                                          
   at OCPP.Core.Management.Program.Main(String[] args) in /var/www/ocpp/data/www/idmcorp.ru/OCPP.Core.Management/Program.cs:line 34

dallmann-consulting, 2021-05-19:
The configuration files currently have endpoints for http and https defined. For https the project contains a self-signed "localhost.pfx" certificate which is referenced in the configuration file:

```
  "Kestrel": {
    "Endpoints": {
      "Http": {
        "Url": "http://localhost:8081"
      },
      "HttpsInlineCertFile": {
        "Url": "https://localhost:8091",
        "Certificate": {
          "Path": "localhost.pfx",
          "Password": "OCPP.Core"
        }
```

Your log says:
Unhandled exception. Interop+Crypto+OpenSslCryptographicException: error:2006D080:BIO routines:BIO_new_file:**_no such file_** at ...

So it doesn't find the referenced certificate file.

vasya00044, 2021-05-19:
Thank you. Should I just rename localhost.pfx to 37.143.8.169.pfx? Or is there something else that needs to be done?

dallmann-consulting, 2021-05-19:
No, that is just a filename. Most probably the certificate or file path must be changed on Linux.  You have to google that...
I would recommend to remove the https endpoints in both projects temporarily. Then check if the application works. If you are working in an internal network for testing purposes you don't require a https connection.  You can fix the certificate problem later.

dallmann-consulting, 2021-05-19:
Have a look at this:
https://stackoverflow.com/questions/59254449/kestrel-certificate-no-such-file-on-linux-containers-works-find-on-windows-con

closed 2021-06-04 by dallmann-consulting

vasya00044, 2021-06-14:
Hi.
Is there an OCPP API for the server? I want to make an application with a
payment system for payment on time and payment for kW.