dallmann-consulting/OCPP.Core #4: SQLite Error 14: 'unable to open database file'.

tianmu88888, 2021-05-28:
Hi,  I have been download form here and try to build this.
But the Web page told me that  "SQLite Error 14: 'unable to open database file' "

I'm using sqlite so I done this change:

  "ConnectionStrings": {
    "SQLite": "Filename=.\\..\\SQLite\\OCPP.Core.sqlite;"
    //"SqlServer": "Server=.;Database=OCPP.Core;Trusted_Connection=True;"
  },
and rename the sql file from OCPP.Core.Empty.sqlite to OCPP.Core.sqlite

But I still get the error?
is there anything I did wrong?

Thanks!

dallmann-consulting, 2021-05-28:
My first guess: you need to escape the backslashes in the path with an extra backslash:
"SQLite": "Filename=.\\..\\SQLite\\OCPP.Core.sqlite;"
If that doesn't help try an absolute path instead of a relative path. If you're running in IIS you also need to make sure that the App-Pool account has permissions on the file.
Also be aware that both projects (Server and Web-UI) need access to the database (see both config files).

tianmu88888, 2021-05-29:
I use all of the default setting except the SQLite.
I change the path first and then run it with  "Build with SDK" this part.
then copy the wwwroot to all the the folder.
I did it twice from the start, but get the same error.
Can I give you more things to get more help?

dallmann-consulting, 2021-05-30:
Please try an absolute file path (like c:\\...myPath...\SQLite\OCPP.Core.sqlite).
The error message says that the database file can not be found or opened. So it can be a wrong path or an access denied problem.
If you need more help  I need more information about your configuration (exact file paths, IIS or Kestrel [simply start the exe files]).

tianmu88888, 2021-05-30:
Hi, I try the absolute path to fix this,But still wrong
And I upload the file I use and the file path for you.
https://drive.google.com/drive/folders/1du9Ed9nW_1xVfnxSAuqfSTn7k-Mi4Lb1?usp=sharing
Here is the link , Sorry for your inconvenient

dallmann-consulting, 2021-05-30:
If I'm not mistaken your build output is somewhere else. The ZIP file only contains the binary of the database project. I can't help you with these files.
Please send exakt screenshots of:
- the exact path to the output directories of the Server and Web-UI.
  It must contain this: https://github.com/dallmann-consulting/OCPP.Core/blob/main/images/BuildOutput.png
- screenshot/path to the database file. You said you renamed the file but the ZIP contains the original file.
- Verify that the "appsettings.json" in both projects (in above output directories) contain the correct absolute path to your database file. Please send screenshots of these as well.

tianmu88888, 2021-06-01:
Sorry I upload the wrong file , Because I'm trying to use the SQL Server at the same time, I have been update the file in the same url

1. 
[image]
[image]
Here is the path picture.
2. I keep a empty sqlite because if I make some mistake I can restart it.
3. 
[image]
[image]
[image]
Here is the SQLite path and the appsetting.json
Thanks for your help

dallmann-consulting, 2021-06-01:
I think I found it. SQLite is a file based database (like Microsoft Access). You have configured the directory path but you need to point to the ".sqlite" file. 
Both ".sql" files are just SQL scripts for creating a new database or migrating from a previous version. You don't need them if you start with the empty sqlite file.

dallmann-consulting, 2021-06-03:
Is it working?

tianmu88888, 2021-06-04:
That works! Thanks!
But I have another question , how can I connect to the real charger?

dallmann-consulting, 2021-06-04:
That's a question to the manufacturer of your charger. The charger connects to the OCPP server and needs to be configured  - it's the same with a simulator. The server is passive and just waits for connection request.

tianmu88888, 2021-06-04:
Yes Our Charger is support with OCPP so I just enter the same setting in the simulator to the charger and then it can works?
And I don't know why my simulator only can connect at the first time and can't send meter value to the server

dallmann-consulting, 2021-06-04:
Yes - basically it's the same with the charger. In some models you need to fill seperate fields (Server/IP, ID etc.)  and in some you enter more or less the URL like in the simulator. You need to read the documentation of your charger. It can complicated because often the chargers don't have a Web-UI and you need to configure it with files on an USB stick.

What do you mean with "can only connect at the first time"? With the OCPP "J" protocol the chargers are always connected.  They have a constant web socket connection.
When the JSON is not accepted it might be a bug in the simulator or the OCPP server. Can you exactly describe what you are doing? Like: 1. Connect; 2: Start transaction; 3. ...

tianmu88888, 2021-06-04:
Thanks I'll try my own charger.

It's my description failed , I mean only my first charge point can be use.
And the ERROR message is from the step:
Connect→Start Transaction→Type 20 in the meter value→Send meter Values→error

dallmann-consulting, 2021-06-04:
What simulator are you using? The ones I modified and added to the OCPP.Core project? I fixed the base simulators in respect to the time as well. The OCPP specification recommands using UTC time. I'm UTC+1/+2 (depending on summer/winter time and it works. This effect can happen when the charger/simulator start the transaction with a **local** timestamp (in your case UTC+8h). When the Web-UI calculates the duration it uses UTC. Then the duration is negative.

dallmann-consulting, 2021-06-04:
Hold on - I get the same problem (JSON is not accepted). There seems to be a problem with the timestamps.

tianmu88888, 2021-06-04:
Yes I think there is some problem to these simulator.
I am using the one you recommend,[OCPP1.6 Simple Chargebox Simulator](https://github.com/victormunoz/OCPP-1.6-Chargebox-Simulator)
So I might be wrong in these days?

dallmann-consulting, 2021-06-04:
I found and fixed it.
It was a problem with the simulators.  The hours in the timestamp were not always two digits (like "8" instead of "08"). I'm ususally working in the evening hours on the project - so I didn't notice the problem. Thank you for living so far away  :-) 

Get the updated simulators: https://github.com/dallmann-consulting/OCPP.Core/tree/main/Simulators

closed 2021-06-20 by dallmann-consulting