dallmann-consulting/OCPP.Core #6: Start Transaction From Remote

BurakOgutken, 2021-07-19:
Hello,

First of all, thank you very much for this beautiful example. Beautifully crafted down to the smallest detail. I have two questions;

1. Can we send open or close transactions to the station via the admin page or api.
2. How do we support 1.5 version.

For my first question, I tried something but failed. I am sharing the code below

thank you
`private async Task BeginTransaction16(ChargePointStatus chargePointStatus, HttpContext apiCallerContext)
        {
            ILogger logger = _logFactory.CreateLogger("OCPPMiddleware.OCPP16");
            ControllerOCPP16 controller16 = new ControllerOCPP16(_configuration, _logFactory, chargePointStatus);

            Messages_OCPP16.StartTransactionRequest startTransaction = new Messages_OCPP16.StartTransactionRequest();
            startTransaction.ConnectorId = 2;
            startTransaction.IdTag = "1";
            startTransaction.Timestamp = DateTimeOffset.Now;
            startTransaction.ReservationId = 0;
            startTransaction.MeterStart = 0;
            string jsonResetRequest = JsonConvert.SerializeObject(startTransaction);

            OCPPMessage msgOut = new OCPPMessage();
            msgOut.MessageType = "2";
            msgOut.Action = "StartTransaction";
            msgOut.UniqueId = Guid.NewGuid().ToString("N");
            msgOut.JsonPayload = jsonResetRequest;
            msgOut.TaskCompletionSource = new TaskCompletionSource<string>();

            // store HttpContext with MsgId for later answer processing (=> send anwer to API caller)
            _requestQueue.Add(msgOut.UniqueId, msgOut);

            // Send OCPP message with optional logging/dump
            await SendOcpp16Message(msgOut, logger, chargePointStatus.WebSocket);

            // Wait for asynchronous chargepoint response and processing
            string apiResult = await msgOut.TaskCompletionSource.Task;

            // 
            apiCallerContext.Response.StatusCode = 200;
            apiCallerContext.Response.ContentType = "application/json";
            await apiCallerContext.Response.WriteAsync(apiResult);
        }`

BurakOgutken, 2021-07-19:
I think I solved it first one, I made a mistake in this part, it worked.

  msgOut.Action = "StartTransaction";
  msgOut.Action = "RemoteStartTransaction";

dallmann-consulting, 2021-07-19:
Hi Burak,

many thanks for your praise. If you like it give it a star :-)

1.)  The OCP-Protocol allows you to remote start and stop transactions with special messages: RemoteStartTransaction and RemoteStopTransaction. I haven't implemented these because you don't really need them in a home usage scenario. But you can easily implement them in the API.

2.) I focused on OCPP 1.6 because it is the most stable and common version.
Wikipedia:
> For now only OCPP 1.6 can be fully tested and certified. OCPP versions older than OCPP 1.6 (such as OCPP 0.7, OCPP 1.2, and OCPP 1.5) as well OCPP 2.0 are not covered under the current OCPP certification program.

According to the OCPP1.6 documentation, it is not backward compatible to 1.5:
> OCPP 1.6 is based on OCPP 1.5, with some new features and a lot of textual improvements,
> clarifications and fixes for all known ambiguities. Due to improvements and new features, OCPP 1.6 is
> not backward compatible with OCPP 1.5.
To support 1.5 you probably need to add a new controller with all messages and actions. I guess you can copy&paste a lot from 1.6 but it still requires implementing & testing all messages.

One more thing:
I invested many hours of work in the project and it is published under GPL. That means that you are supposed to publish your changes and extensions as well. I would like to ask you to note and respect this.
If you just want to add things and not reengineering everthing we can integrate it here.

closed 2021-08-31 by dallmann-consulting