dallmann-consulting/OCPP.Core #7: Cant connect with simulator

jpaquete, 2021-07-28:
Hello, I am trying to connect the simulator at localhost without success (of course it is my bad, but I hope you can give any clues) 
Server is running at localhost:8081 tag and station id are created

WebSocket connection to 'ws://localhost:8081/OCPP/station42' failed: WebSocket is closed before the connection is established.
wsConnect @ simple simulator1.6.html:128
(anonymous) @ simple simulator1.6.html:336
dispatch @ jquery-3.3.1.min.js:2
y.handle @ jquery-3.3.1.min.js:2
simple simulator1.6.html:239 ws connection error: 1006

dallmann-consulting, 2021-07-29:
It's hard to help you without any details...
- Which platform are you using? How do start the server? What database are you using?
- What happens when you open "http//localhost:8081" in a browser? You should see a simple message that the server is running and how many charge points are connected.
- My guess is that the paths (log and dump) are not correct which leads to errors. Otherwise the server should write a log file. That should be helpful.

jpaquete, 2021-07-29:
Hello Ulrich, I am using an instance of sql server express 2012. Aperently with no problems  on the frontend because I was able to login and register a new charger and a new tag (cant see any log C:\temp\OCPP) Please check attached image with all info. Thanks !

[image]

dallmann-consulting, 2021-08-01:
Sorry for the delay...
Your URL is missing the "OCPP" part. It must be:

ws://localhost:8081/**OCPP**/station42

dallmann-consulting, 2021-08-02:
I just noticed that you already tried the URL with OCPP. 
Can you try the original path for log files? It should be a relative path and you should see a "log" directory.

closed 2021-08-31 by dallmann-consulting