dallmann-consulting/OCPP.Core #10: please help me out

dineshkumar99999, 2022-03-21:
Hello I have created charge points and server page using ocpp.core.server and ocpp.core.management. Using OCPP1.6 simulator I can able to connect, start, stop charging. But now my intention is Schedule charging. From ocpp2.0 simulator when I am giving start,stop,schedule it shows JSON  is not accepted.
[image]

please help me out to solve my doubts

dallmann-consulting, 2022-03-22:
Hi,
it is my understanding that the OCPP backend can send a schedule to the charging station. 
Are you trying to send a schedule from the charging station (=simulator) to the server? The OCPP messages don't allow that and it doesn't seem to make sense to me because the charging station is the one to adhere to the schedule. Is this a misunderstanding?
Can you attach the exakt message you are sending?

dineshkumar99999, 2022-03-24:
Yes OCPP backend can send a schedule to the charging station. I am also trying the same. But it is not working. when I click on Notify EV charging Schedule on cp20 simulator it is not working

dallmann-consulting, 2022-03-26:
You mean the "Notify EV Charging Schedule", correct? This works for me:

[image]

The simulator sends a 'NotifyEVChargingSchedule'  message and the OCPP server answers with status=accepted.
Are you using the simulators included in this project?

dineshkumar99999, 2022-03-29:
Yes I am using the simulators which is included in this project. Then why it is not working for me?

dallmann-consulting, 2022-03-29:
Sorry - without further details (ocpp messages and log) I cannot help you.

dallmann-consulting, 2022-04-19:
No further activity... close thread.

closed 2022-04-19 by dallmann-consulting