dallmann-consulting/OCPP.Core #11: Simulator

billywatsy, 2022-04-14:
l have cloned the project fine, 

if l use the simulator and request connection, it always fails, debugging the code l have seen that it requires a basic auth token ( username and password ). l have done the steps of setting the charge point via the management : OCPP.Core.Management

The question is more on the simulator on why it's not sending the token in the header or where is it set on the simulator, l have used the latest in your git. or l am doing something wrong 

[image]

2:
[image]
here is where it's failing , so as to get the first connection thanks

billywatsy, 2022-04-18:
l have seen how to set it 

http://username:password@example.com/

closed 2022-04-18 by billywatsy

dallmann-consulting, 2022-04-19:
If your are just using the simulator, you don't need an authentication. You probably have configured a username/password in the chargepoint configuration. That's why the server requires the authentiation. Just clear these fields and it should work without.