dallmann-consulting/OCPP.Core #14: Real Charger testing

rahulbhat13, 2022-06-16:
@dallmann-consulting Does it work with Real charger -> can you suggest me the details any modification I need to take care of .

Since https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/resources/config/prod/main.properties#L9-L13 works fine with real charger , is it can be done ?

dallmann-consulting, 2022-06-16:
What gave you the impression that it doesn't work with real chargers?
As statet in the base page: we are using it in our multi family home with currently 6 chargers (Keba P30). And others seem to use it as well with real chargers. 
But no software is error free - that also includes the chargers!

rahulbhat13, 2022-07-01:
Ok sure , since wanted to check whether we can implement with multiple chargers and register it accordingly and even connect to the Mobile app

closed 2022-07-25 by dallmann-consulting

rahulbhat13, 2022-08-16:
It was not working with multiple charger at the same time . Do you have any idea or details we want to setup multiple charger  and also load testing along with it