dallmann-consulting/OCPP.Core #16: Start Transaction Error

receptanilcengiz, 2022-07-10:
Hello

I am testing server with INFINITY Simple OCPP 1.6 Chargebox Simulator provided in Readme file (https://github.com/victormunoz/OCPP-1.6-Chargebox-Simulator) 

when I click Start Transaction button it gives this error
fail: OCPPMiddleware.OCPP16[0]
      OCPPMiddleware.Receive16 => Unknown message type: 1 / Msg: [1,"ETPCN52r6YfYsB0xI2G2imfhjDf3RoUVoVe9","StartTransaction",{"connectorId":1,"idTag":"12345678","timestamp":"2022-07-10T3:11:55Z","meterStart":0,"reservationId":0}]

How can I solve this?

billywatsy, 2022-07-11:
hi @receptanilcengiz  did you first manage to connect first ?

connect-> authorize -> start tranx ?

receptanilcengiz, 2022-07-11:
yes I do with these orders

dallmann-consulting, 2022-07-11:
Use a better simulator. Message type "1" is not defined in specification and therefore invalid. Normal msg types are 2/3 and 4 means error.

Bernd81, 2022-07-12:
Hi @receptanilcengiz 
did you try to restart your Wallbox?
I had a similar issue with my Alfen Eve Single S-line wallbox. After reinitializing with a restart through management software of the box everything is working well.

dallmann-consulting, 2022-07-14:
That one should work. Did you also try the modified one included in this repository?
Nevertheless, your error message states unknown message type "1" which is truly not allowed. 
Try different simulators and see what happens.

dallmann-consulting, 2022-07-25:
No more activities

closed 2022-07-25 by dallmann-consulting