dallmann-consulting/OCPP.Core #17: Data exchange failed - JSON in not accepted

stig888881, 2022-07-16:
Good afternoon, I encountered such a problem, when using OCPP-2.0-CP-Simulator, it gives such a message, in the server logs it writes the following:
fail: OCPP.Core.Server.ControllerOCPP20[0] @ 2022-07-16T12:59:35.1770301+03:00
TransactionEvent => Exception: Required property 'transactionInfo' not found in JSON. Path '', line 1, position 475.
Newtonsoft.Json.JsonSerializationException: Required property 'transactionInfo' not found in JSON. Path '', line 1, position 475.
at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.EndProcessProperty(Object newObject, JsonReader reader, JsonObjectContract contract, Int32 initialDepth, JsonProperty property, PropertyPresence presence, Boolean setDefaultValue)
at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
at Newtonsoft.Json.JsonSerializer.Deserialize(JsonReader reader, Type objectType)
at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value)
at OCPP.Core.Server.ControllerOCPP20.HandleTransactionEvent(OCPPMessage msgIn, OCPPMessage msgOut) in C:\Users\ulrichd\source\repos\GitHub\OCPP.Core\OCPP.Core.Server\ControllerOCPP20.TransactionEvent.cs:line 46
[image]

dallmann-consulting, 2022-07-25:
Can you repro this error and what exactly are you doing?
The dump shows that the error is thrown in a transaction event (start/stop/update). A mandatory transaction information is missing in the message which is a client problem (=simulator) problem.
When using these simple simulators you must adhere to certain rules (like start transaction before update or stop). Otherwise you will get errors.

dallmann-consulting, 2022-08-22:
No more activity

closed 2022-08-22 by dallmann-consulting