dallmann-consulting/OCPP.Core #18: How to connect to device

MichalMu, 2022-08-05:
Hello, I have  a problem to connect the device to the server. Could you explain how to do it?
Also, how can I connect by simulator using IP addres?
For example when I put in Central station:   ws://localhost:8081/OCPP/station42, everything is ok, but when I put ip addres of local device it doesn't work: ws://192.168.1.2:8081/OCPP/station42

billywatsy, 2022-08-05:
can you try 
ws://Test1234:Test1234@xxxxxxxxxx/OCPP/Test1234

for basic auth

dallmann-consulting, 2022-08-05:
It's a firwall problem. With "localhost" you are using a (local) shortcut in your network stack. When you use a regular IP-address (even your own) your calls need to pass the firewall. 
You need to open the corresponding port in your firewall.

dallmann-consulting, 2022-08-22:
No more activity

closed 2022-08-22 by dallmann-consulting