dallmann-consulting/OCPP.Core #19: Example of how to send a message from server to client

profnimrod, 2022-08-09:
Could you please provide an example of how to send a message from the server to a ChargePoint using the library.  Say, for example, a SetChargingProfile message.  There doesn't seem to be a way via the UI, but it looks like the library supports it.

billywatsy, 2022-08-16:
hi @profnimrod 

you can expose what you want to send via API and let the UI call that endpoint , l haven't checked the UI only the server but @dallmann-consulting can advice 

but want l think is expose the functionality via API like what was done on the reset , you can post your values as body or query params of the properties you need in the chargeprofile and have a method which formats those to SetChargeProfile

for example check how reset is done 
[image]

calling the endpoint /API/Reset/{chargepointId} 
[this is a call on the API and will then send to the chargepoint a reset command if charge is online]

so in this case   you can have 
calling the endpoint /API/SetChargeProfile/{chargepointId}   
and have also another method to handle response 

Hope this helps

dallmann-consulting, 2022-08-22:
Thank you @billywatsy. Absolutely correct :-)

You need to implement the message in the server API. The UI uses the server's API to send a reset or unlock message. The server blocks your call until it receives an answer from the charger and sends you a result. So from an API perspective it seems like a synchronous call.
To set a charging profile with a power limit was not in my focus for the first version. It would make sense with the semi dynamic power management described in issue #21.

profnimrod, 2022-08-22:
Thanks for your comments guys - I'll take a crack at a charge profile message and let you know how I get on.

profnimrod, 2022-08-24:
I've nearly got this working now... I'm afraid I don't have the time to detail my approach (it is based on guidance above), but I have included a link the full updated VS Solution for both the server and management tool in case anyone is interested.  At the moment a hard-coded profile is pushed out, so that'll need to change, but on the whole I think it is close to what is needed.  At the moment my test charger is rejecting the message, but I suspect I just have a minor bug somewhere.

[VS Solution](https://inergysystems.blob.core.windows.net/temp/InSysOCPPServer.7z?sv=2021-04-10&st=2022-08-23T23%3A49%3A32Z&se=2023-08-24T23%3A49%3A00Z&sr=b&sp=r&sig=ARM6In%2BNql%2BZI54W%2Fk6fvggFmsjMBzf%2B1P2KEINLELo%3D)

profnimrod, 2022-10-11:
HI

The example of a Reset is what is referred to as an “operation initiated by charge point” in the OCPP 1.6 standard, i.e., device request and server response.  Most of the 10 charge point operations are included in OCPP.Core as far as I can tell.

The SetChargingProfile is an example of an “operation initiated by central system”, i.e., server request and device response – the other way round… There are 19 central system operations, including SetChargingProfile, and as far as I can tell none of them are included in OCPP.Core.

I don’t think it should be too hard to add the 19 “operations initiated by central system” in OCPP.Core.  I think all that has to be done is to cache the web socket for each charge point when the connection is created and then create the SetChgargingProfile message and send to the Chargep point via:

await SendOcpp16Message(msgOut, logger, chargePointStatus.WebSocket)

Something like that.  It would be great to have the dev’s recommendation on how to add these additional messages to the library.  I’m happy to help code them up.

Thanks, Kurt

[image]<https://user-images.githubusercontent.com/22769210/184964006-14760b01-7dd2-44c2-93ab-ef0aba580d71.png>

Manojk-Singh, 2024-06-03:
> I've nearly got this working now... I'm afraid I don't have the time to detail my approach (it is based on guidance above), but I have included a link the full updated VS Solution for both the [...]

Hi, vs solution download link is not working.

dallmann-consulting, 2024-09-17:
SetChargingProfile is implemented now and the usage (API) is documented.
So I'm closing the ticket.

closed 2024-09-17 by dallmann-consulting