dallmann-consulting/OCPP.Core #20: Energy Limit on a transaction - Advice

billywatsy, 2022-08-16:
hi All

Is it possible to send to the OCCP client using the energy value, after which the charger will automatically stop charging?

anyways to which might work 

Here is what l think

1: Charge profiles -
{
"connectorId": 0,
"idTag": "xxxxxxxxxxxx",
"chargingProfile": {
"chargingProfileId": 10,
"transactionId": null,
"stackLevel": 0,
"chargingProfilePurpose": "TxProfile",
"chargingProfileKind": "Relative",
"recurrencyKind": "Daily",
"validFrom": "2022-02-18T10:18:46.000Z",
"validTo": "2022-08-18T10:18:46.000Z",
"chargingSchedule": {
"duration": 4200,
"startSchedule": "2022-08-18T10:18:46.000Z",
"chargingRateUnit": "W",
"chargingSchedulePeriod": [
{
"startPeriod": 0,
"limit": 50
},
{
"startPeriod": 120,
"limit": 0
}
]
}
}
}

The last one will send zero currents and probably that time will stop

2: Custom Data Transfer  , but this is more of confirming with the vendor right?

profnimrod, 2022-08-16:
This is essentially the same as https://github.com/dallmann-consulting/OCPP.Core/issues/19 - basically how do we do server -> client messaging?

dallmann-consulting, 2024-10-15:
ChargingProfiles are implemented now. See API description.

closed 2024-10-15 by dallmann-consulting