dallmann-consulting/OCPP.Core #22: Connect over IP

marsisyazilim, 2022-10-14:
Hi 

I can make it work using localhost address (ws://localhost:8081/OCPP/TestAAA) but over the IP (ws://192.168.1.116:8081/OCPP/TestAAA) doesnt work. what setting should I do? I added rule for firewall 8081 port. still not working.

closed 2023-05-03 by dallmann-consulting

xsherlockpl, 2024-10-06:
I have exactly the same issue and quite don't know how to enable the webserver to be accessible over IP. My VM Ububntu 20.04 is not running any firewall and still the service is only open on the localhost. 

`sherlock@ubuntutest:~$ nmap localhost
Starting Nmap 7.80 ( https://nmap.org ) at 2024-10-06 21:14 CEST
Nmap scan report for localhost (127.0.0.1)
Host is up (0.000087s latency).
Not shown: 995 closed ports
PORT     STATE SERVICE
22/tcp   open  ssh
139/tcp  open  netbios-ssn
445/tcp  open  microsoft-ds
8081/tcp open  blackice-icecap
8082/tcp open  blackice-alerts

Nmap done: 1 IP address (1 host up) scanned in 0.07 seconds
sherlock@ubuntutest:~$ nmap 172.16.10.104
Starting Nmap 7.80 ( https://nmap.org ) at 2024-10-06 21:15 CEST
Nmap scan report for 172.16.10.104 (172.16.10.104)
Host is up (0.000097s latency).
Not shown: 997 closed ports
PORT    STATE SERVICE
22/tcp  open  ssh
139/tcp open  netbios-ssn
445/tcp open  microsoft-ds

Nmap done: 1 IP address (1 host up) scanned in 0.06 seconds
sherlock@ubuntutest:~$ sudo ufw status
[sudo] password for sherlock:
Status: inactive
`

mstrini, 2024-10-06:
In both projects, `OCPP.Core.Management` and `OCPP.Core.Server` is a file `appsettings.json` and there are endpoints defined.
You have to change the binding from "localhost" to something different like "0.0.0.0" to allow other clients than the local machine to access the endpoints.

xsherlockpl, 2024-10-06:
Oh great, I tried the actual IP of this machine and that does not work as it enables the managment but breaks server talking to managment, It must be 0.0.0.0
Thanks