dallmann-consulting/OCPP.Core #32: Possible DoS Vulnerability: No Control on length of 'chargePointVendor' property in 'BootNotification' OCPP message

Tano-Coppoletta, 2023-11-02:
**Description of the possible vulnerability**: 
While performing black box testing on OCPP.Core on my local machine, I discovered a potential security vulnerability that might lead to a Denial of Service (DoS) attack.

**Details**:
- There appears to be no control over the length of the `chargePointVendor` property of the `BootNotification` OCPP message. According to the protocol specifications, the `maxLength` of this property should be 20 characters.
- I conducted a test by sending a `BootNotification` message with a `chargePointVendor` length of 442 MB, resulting in the server crashing.

**Attachments**:
- I have attached a screenshot of the server logs that displays the behavior when the oversized `BootNotification` message was received.

[image]

dallmann-consulting, 2023-11-02:
Hi Gaetano,

thank you for your work and the notification!

I'm not sure what exactly causes the server to crash. I would expect that with such a large message the deserialization would simply fail with an exception:

```
            try
            {
                Logger.LogTrace("Processing boot notification...");
                BootNotificationRequest bootNotificationRequest = JsonConvert.DeserializeObject<BootNotificationRequest>(msgIn.JsonPayload);

           ...

            }
            catch (Exception exp)
            {
                Logger.LogError(exp, "BootNotification => Exception: {0}", exp.Message);
                errorCode = ErrorCodes.FormationViolation;
            }
```

I don't see a realistic way to check specific fields because the whole message gets deserialized at once. And OCPP.Core doesn't have a problem with a too long vendor information (and other fields). 
A simple solution could be to define a max. size for incoming messages and check that. The ocpp messages have all small sizes and a configurable limit of 1-2 MB should be enough for all messages.

In general OCPP.Core is intended for use in a protected environment/network. For a public reachable API there should be more precautions. And the existing authentication mechanismens should be used as well and can avoid many attacks.

Tano-Coppoletta, 2023-11-06:
Dear dalmann-consulting,

Thank you for addressing my previous concerns. I have observed that while the 'BootNotification' with an oversized 'chargePointVendor' field leads to a Denial of Service (DoS) attack by crashing the server, there is a broader issue at hand. Oversized 'idTag' values in 'Authorize' messages, although smaller in size (e.g., 10 MB), are being accepted and stored in the 'MessageLog' table of the database. This storage of excessively large entries without prior validation could be exploited to fill the database with large, non-standard data, potentially impacting performance and storage.

It seems prudent to consider implementing a pre-validation step to check message sizes against the protocol defined threshold before they are parsed and logged, ensuring database integrity and preventing possible storage abuse.

Kind regards,
Gaetano Coppoletta

dallmann-consulting, 2023-11-13:
Please check with V1.2.0 and enabled message validation (and max. message size)

Tano-Coppoletta, 2023-11-17:
It works now, it's not accepting long messages and so the server is not crashing anymore