dallmann-consulting/OCPP.Core #33: Possible Inconsistency in Transaction Meter Values Allowing Negative Charging

Tano-Coppoletta, 2023-11-06:
**Description of the Issue**:
I have discovered a possible logic flaw in transaction handling. Specifically, the server accepts `StopTransaction` messages where the `meterStop` value is lower than the `meterStart` value from a corresponding `StartTransaction` message, leading to the storage of a negative charged value in the database.

**Steps to Reproduce**:
1. Send a `StartTransaction` message with a `meterStart` value.
2. Send a `StopTransaction` message with a `meterStop` value lower than the `meterStart` value.
3. Observe that the server processes the transaction and stores a negative value for the charged amount.

**Expected Behavior**:
The server should validate that the `meterStop` value is greater than or equal to the `meterStart` value to ensure transaction consistency and prevent illogical charging data.

**Actual Behavior**:
The server stores the transaction with a negative charged amount, which is illogical in the context of energy metering and could lead to financial discrepancies.

**Potential Impact**:
If left unaddressed, this could result in incorrect billing calculations, impacting the integrity of transactional data and potentially causing financial losses or disputes.

**Attachments**:
- A screenshot showing the negative charged amount as recorded in the system: 
- [image]
[image]

**Suggested Solution**:
Implementing server-side validation checks to confirm that `meterStop` is not less than `meterStart` before accepting and storing transaction data would mitigate this issue.

Thank you for your attention to this matter. Looking forward to your response.

Best regards,
Gaetano Coppoletta.

dallmann-consulting, 2023-12-06:
I'm planning to add an optional config switch for that.
OCPP.Core has no integrated billing. So there is no real risk of negative payments. 

With Issue #31 we have seen that the server should not simply reject a StopTransaction-Request. This might lead to problems in the chargers state because the car can always force a stop. So when the charger sends a lower meter value the server should still accept the stop request. It is the mistake of the charger if it sends a wrong meter value. But rejecting the stop request most probably means that the charger repeats sending the same stop request over and over.

That's why I'm planning to add an **OPTIONAL** config switch for that.

closed 2025-01-12 by Tano-Coppoletta

dallmann-consulting, 2025-01-14:
That is a very good question and I can't answer it. But it seems logical.
I know that the upcoming OCPP 2.1 standard will include scenarios like V2X. The specification hasn't been released so far.