dallmann-consulting/OCPP.Core #34: `StopTransaction`: a Transaction Can Be Stopped with Random Transaction ID

Tano-Coppoletta, 2023-11-06:
**Issue Description**:
The server allows a `StopTransaction` message to stop an active transaction using a random `transactionId` instead of requiring the specific `transactionId` assigned at the start of the transaction. This could potentially lead to unauthorized stopping of transactions.

**Steps to Reproduce**:
1. Start a transaction using `StartTransaction`, receiving a valid `transactionId`.
2. Send a `StopTransaction` message with a random `transactionId` instead of the one received in step 1.
3. Observe that the transaction is stopped by the server.

**Expected Behavior**:
The `StopTransaction` message should be validated to ensure it includes the correct `transactionId` that was assigned at the start of the transaction, thereby binding the user to the specific transaction being stopped.

**Actual Behavior**:
The server stops an active transaction when a `StopTransaction` message is sent with any random `transactionId`.

**Suggested Solution**:
Implement strict validation on the server to only stop transactions when the `StopTransaction` message contains the correct `transactionId` provided at the beginning of the transaction.

I look forward to your input on this matter.

Best regards,
Gaetano Coppoletta

dallmann-consulting, 2023-11-16:
You are right - it was initially by design. I didn't think of multiple connectors and wanted the system to be fault tolerant. If the charger sends an unknown transcation id, the server checks if the last transaction (for that charger!) is not closed and uses that.
But that behavior is dangerous with multiple connectors and should be removed.
The next version (1.3) will check the transaction id strictly.

dallmann-consulting, 2023-11-30:
Please see V1.3.0.
The issue should be fixed there.