dallmann-consulting/OCPP.Core #36: Unvalidated Acceptance of Additional and Duplicate Properties in `StartTransaction` Message

Tano-Coppoletta, 2023-11-06:
**Issue Description**:
The server accepts additional, arbitrary properties in the `StartTransaction` message, and when duplicate properties are provided, it accepts the last occurrence without validation. For example, a message with two different `connectorId` properties is processed using the value of the last one.

**Steps to Reproduce**:
1. Send a `StartTransaction` message with a custom additional property.
2. Send a `StartTransaction` message with duplicate properties, such as two different `connectorId` values.
3. Observe that the server accepts the message and processes it based on the last occurrence of the duplicate property.

**Expected Behavior**:
The server should either reject messages with unknown additional properties. For duplicate properties, it should reject the message.

**Actual Behavior**:
The server stores the value from the last occurrence of a duplicate property in the `StartTransaction` message, leading to potential misconfigurations.

**Potential Impact**:
Accepting additional and duplicate properties without validation may lead to unpredictable system behavior and could be exploited to alter transaction records or impact system integrity.

Best regards,
Gaetano Coppoletta

dallmann-consulting, 2023-11-13:
Please check with V1.2.0 and enabled message validation

Tano-Coppoletta, 2023-11-13:
I have downloaded the new version but I am not able to send correct messages, I keep receiving a "FormationViolation" error.
Specifically I have sent this three messages
[2,"1","Heartbeat",{}]
[2,"1","Authorize",{"idTag":"1"}]
[2, "1", "StartTransaction", {"connectorId": 1, "idTag": "1", "meterStart": 2, "reservationId": 11, "timestamp": "2023-10-03T12:34:56Z"}]

and I have received this three responses:
[3,"1",{"currentTime":"2023-11-13T18:10:56.356693+00:00"}]
[4,"1","FormationViolation","",{}]
[4,"1","FormationViolation","",{}]

dallmann-consulting, 2023-11-13:
You asked for it  :-)

Please look in the log files. The validation errors are shown there. A working Authorize message is:
[2,"2ToBmpHjfJn7lhkHxQe6hLV6KbASiYjLimHd","Authorize",{"idTag":"B4A63CDF"}]

Tano-Coppoletta, 2023-11-13:
It's not working for me, maybe I am missing something but I am sending this message:
[2,"2ToBmpHjfJn7lhkHxQe6hLV6KbASiYjLimHd","Authorize",{"idTag":"B4A63CDF"}]
and I am still getting this:
[4,"2ToBmpHjfJn7lhkHxQe6hLV6KbASiYjLimHd","FormationViolation","",{}]
The log says:
[image]

And in the MessageLog table I have this records:
[image]

dallmann-consulting, 2023-11-14:
The log was helpfull: this seems to be a platform issue. You are working on Linux, right?
The code searches for a matching schema file and determines the code location (directory). In Windows this seems to be a URI (file://...) and in linux it seems to be different.
If you are working with the source code I could send you a (hopefully) corrected version to check.

Tano-Coppoletta, 2023-11-14:
Yes, I am working on Linux. I am doing black box testing so I have not been through the source code. The previous version was working, this time I had to change the path in appsettings.json to substitute \\ with /. It's curios that it works with an Heartbeat message but not with the others.

dallmann-consulting, 2023-11-14:
Can you try to replace the attached files?
[OCPP.Core.Server.zip](https://github.com/dallmann-consulting/OCPP.Core/files/13355844/OCPP.Core.Server.zip)

Hopefully this resolves the problem. If it works, I will publish a new version.

Tano-Coppoletta, 2023-11-17:
Yes, it works.
I am still able to send messages with the same message id. e.g.
[2,"1","Authorize",{"idTag":"B4A63CDF"}]
[2,"1","StartTransaction", {"connectorId": 1, "idTag": "1", "meterStart": 2, "reservationId": 11, "timestamp": "2023-10-03T12:34:56Z"}].

Moreover, I am still able to add additional properties and\or the same property more than once in the same message. This message is still accepted:
[2,"1","StartTransaction", {"connectorId": 1, "idTag": "1", "meterStart": 2, "reservationId": 11, "timestamp": "2023-10-03T12:34:56Z","meterStart":444, "aa":1}]
response: [3,"1",{"idTagInfo":{"expiryDate":"2199-12-31T00:00:00-06:00","status":"Accepted"},"transactionId":5}]

dallmann-consulting, 2023-11-18:
**1.) additional properties**
I found that two schema files for Start- and StopTransaction are missing in the GitHub Version. I must have mixed up a folder while cleaning up. I thought I was very careful in that respect - but obviously still not carefull enough. Sorry!

Can you please add these files in the Schema16 folder and try again?
[StartTransactionRequest.json](https://github.com/dallmann-consulting/OCPP.Core/files/13400887/StartTransactionRequest.json)
[StopTransactionRequest.json](https://github.com/dallmann-consulting/OCPP.Core/files/13400888/StopTransactionRequest.json)

**2. Identical message IDs**
This is (currently) by design. The server takes the ID of an incoming message and uses that for the answer. It doesn't need to store or check these IDs any further.
It is my understanding that the message IDs are important for the sender of messages to match the corresponding answers. If the charger uses the same ID again it is its own problem to match the messages.

Thanks for your work!

Tano-Coppoletta, 2023-12-02:
Thank you for your response!
The identical message IDs, in my opinion, could lead to inconsistencies if there is an attacker. 

We are in the process of submitting Common Vulnerabilities and Exposures (CVEs) related to this matter. However, we can wait if you are in the process of fixing it.

Additionally, we intend to publish our findings in an academic paper.

Thank you for the attention to this matter.

dallmann-consulting, 2023-12-06:
Hi,
I still don't see how identical message could lead to inconsistencies or vulnerabiltyies. The server doesn't need them. They might be important for the client to match requests and responses. But there is no point for the server to store and check them. 
And thinking further: faked messages could "burn" message IDs and lead to problems with real messages. So checking and forcing unique message IDs could lead to inconsistencies/problems.
Do you agree?

closed 2023-12-28 by dallmann-consulting