dallmann-consulting/OCPP.Core #37: Web Interface Fails to Display Multiple Transactions Across Different Connector IDs

Tano-Coppoletta, 2023-11-06:
**Issue Description**:
The web interface does not display all active transactions when multiple transactions are started across different connectors. Transactions initiated on one connector are visible, but subsequent transactions on different connectors do not appear in the interface.

**Steps to Reproduce**:
1. Send a `StartTransaction` message using `connectorId` 1 and note the transaction on the web interface.
2. Send another `StartTransaction` message with `connectorId` 2, identical in all other aspects.
3. Observe that the second transaction is not visible in the web interface.

**Expected Behavior**:
The web interface should display all active transactions regardless of the connector ID, provided each transaction is distinct and properly initiated.

**Actual Behavior**:
Only the transaction for the first `connectorId` is displayed on the web interface. Subsequent transactions with different `connectorIds` are not shown.

**Potential Impact**:
This could lead to a lack of visibility for station operators, causing confusion and potential mismanagement of charging stations.

I hope this report helps in pinpointing the issue for a swift resolution.

Best regards,
Gaetano Coppoletta

dallmann-consulting, 2023-11-13:
I can't reproduce the issue. We are using the system in a productive enviroment and our chargers behave like one chargepoint with multiple connectors. And the web UI shows all running transaction correctly. 
Did you refresh the UI? There is currently no automatic refresh. The reason could also be broken transactions in the database from previous tests.

dallmann-consulting, 2023-11-16:
I've extended the 1.6 simulator to handle two connectors. Both transactions are handled correctly on the server side and the WebUI shows the correct status of both connectors. It just needs a manual refresh in the browser.

[image]

Can you please double check the behavior?

Tano-Coppoletta, 2023-11-18:
In the version that you have sent me for linux, I still see the issue