dallmann-consulting/OCPP.Core #39: Error details

WesleyWu001, 2023-12-14:
Hello, we followed the normal process and steps, but when we opened the [http://localhost:8082/](url) ,we saw these error messages displayed on the webpage. Could you please advise on how to resolve this? 
[image]

dallmann-consulting, 2023-12-14:
You configured OCPP.Core to use a SQL-Server but the connection fails. Please check your configuration.
If you don't want to install and use SQL-Server you can use SqlLite (file based database).

WesleyWu001, 2023-12-14:
Thank you very much. We downloaded SQL-Server according to your suggestion, but now a new problem has arisen. How should we solve it?
[image]

dallmann-consulting, 2023-12-14:
There ist still something wrong with the database. Did you create the database with the script and can connect with SQL Management Studio?  It is not enough just to install the SQL Server.
It's probably easier to start with SqlLite. Just use the preconfigured file.

dallmann-consulting, 2023-12-22:
Does it work now?

WesleyWu001, 2023-12-28:
Sorry for the late reply, we have successfully run the ocpp.core and tried to connect to the charge point. If there are any problems, we will report back. Thank you.

closed 2023-12-28 by dallmann-consulting