dallmann-consulting/OCPP.Core #41: How to read current and temperature

SHIUBING, 2023-12-27:
Hello, I would like to know how to collect current and temperature data, and how to display them on the transaction list. Which files do I need to modify?

dallmann-consulting, 2023-12-27:
It all depends on what the charger sends. You can look at the schema of the MeterValues-Reqest to see what measured values are technically **possible**:
https://github.com/dallmann-consulting/OCPP.Core/blob/main/OCPP.Core.Server/Schema16/MeterValuesRequest.json

And here is sample MeterValueRequest from our real life chargers:

`[2,"eb07f294-f7b3-4a70-b2fe-a1ad375236ba","MeterValues",{"connectorId":4,"transactionId":14,"meterValue":[{"timestamp":"2021-04-08T17:40:33.000Z","sampledValue":[{"value":"164.5","context":"Sample.Periodic","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"}]}]}]`

You can see that there is no current, temperature and not even a SoC. Just an updated meter value in Wh.

closed 2023-12-28 by dallmann-consulting

SHIUBING, 2024-01-05:
Where can I see these things?"[2,"eb07f294-f7b3-4a70-b2fe-a1ad375236ba","MeterValues",{"connectorId":4,"transactionId":14,"meterValue":[{"timestamp":"2021-04-08T17:40:33.000Z","sampledValue":[{"value":"164.5","context":"Sample.Periodic","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"}]}]}]"

dallmann-consulting, 2024-01-05:
You can read the OCPP specification for that. But it might help to just format the message:

```
[2,"eb07f294-f7b3-4a70-b2fe-a1ad375236ba","MeterValues",
{
  "connectorId":4,
  "transactionId":14,
  "meterValue":[{"timestamp":"2021-04-08T17:40:33.000Z",
                 "sampledValue":[{"value":"164.5",
				"context":"Sample.Periodic",
				"format":"Raw",
				"measurand":"Energy.Active.Import.Register",
				"location":"Outlet",
				"unit":"Wh"}]}]
}]
```

The two most significant values are:
   "value":"164.5"
   "unit":"Wh"
So the charger sends a new meter value of 164,5 Wh (the charger was brand new...). OCPP also allows other values (e.g. SoC) but our charger doesn't send anything.

dallmann-consulting, 2024-01-22:
This is the raw OCPP message the charger sends to the OCPP backend. I don't think there is a program to "see" this. It is basically readable text in a technical JSON format.

PulsarFX, 2024-03-04:
> You can read the OCPP specification for that. But it might help to just format the message:
>
> ```
> [2,"eb07f294-f7b3-4a70-b2fe-a1ad375236ba","MeterValues",
> {
>   "connectorId":4,
>   "transactionId":14, [...]

Type2 doesn't transfer SoC to the charging station per definition. So the wallbox can't transfer it to the OCPP server. 
I think SoC is added for DC wallboxes with CCS plugs.