dallmann-consulting/OCPP.Core #43: Problems with installing new charging stations.

SHIUBING, 2024-01-03:
I added a new charging station through the management web interface. The charging station ID is L2-U2. However, the following error message appeared in OCPP.Core.Server.exe:
info: Microsoft.AspNetCore.Hosting.Diagnostics[1]
      Request starting HTTP/1.1 GET https://192.168.50.104/OCPP/L2-U2 - -
info: OCPPMiddleware[0]
      OCPPMiddleware => Connection request with chargepoint identifier = 'L2-U2'
info: OCPPMiddleware[0]
      OCPPMiddleware => SUCCESS: Found chargepoint with identifier=L2-U2
info: OCPPMiddleware[0]
      OCPPMiddleware => No authentication for chargepoint 'L2-U2' configured
warn: OCPPMiddleware[0]
      OCPPMiddleware => No supported sub-protocol in '' from charge station 'L2-U2'
info: Microsoft.AspNetCore.Hosting.Diagnostics[2]
      Request finished HTTP/1.1 GET https://192.168.50.104/OCPP/L2-U2 - - - 400 0 - 1.5490ms
How can I resolve this issue?

dallmann-consulting, 2024-01-03:
It looks as if the charger doesn't send the supported protocol versions. When a charger connects it needs to send the supported protocol versions and the server picks one of these. Here is the connect from the 1.6 simulator:
`            _websocket = new WebSocket(wsurl + "" + CP, ["ocpp1.6", "ocpp1.5"]);`

The above log states that an empty string was sent by the charger:
`           No supported sub-protocol in '' from charge station 'L2-U2'`

SHIUBING, 2024-01-04:
Can you explain why the charging station did not send the supported protocol version, even though I confirmed that it supports OCPP1.6JSON?

dallmann-consulting, 2024-01-04:
Please ask the maker of your charger. The OCPP specification states that the charger MUST send a protocol version:

[image]

SHIUBING, 2024-01-05:
OK, thanks