dallmann-consulting/OCPP.Core #44: Start Charge Invalid error

DgnOnur, 2024-01-08:
Hello,
I start charging. It provides energy, but charging stops with an invalid error. Can you share an example of starting and stopping charging?

dallmann-consulting, 2024-01-09:
What invalid error?
You normally start the transaction at the charger. If you want to start it from remote you need to implement the RemoteStartTransaction-Messages. I think Daniel Nord has implemented this in his branch: https://github.com/danielnord/OCPP.Core

DgnOnur, 2024-01-10:
Daniel Nord I was able to perform the operation on the link. the problem persists.

string apiResult = await msgOut.TaskCompletionSource.Task; In this part, you wait for a long time and there is no response.

In Daniel Nord's codes, revise the charging start function content as in the link https://github.com/dallmann-consulting/OCPP.Core/issues/6 and add msgOut.Action = "RemoteStartTransaction"; I updated the part and tried it.

The charging process starts and then gives an error and ends.

LogId	LogTime	ChargePointId	ConnectorId	Message	Result	ErrorCode
1285	2024-01-10 10:07:03.5220623	TACW2242222T1841	1	StartTransaction	Invalid	NULL
1286	2024-01-10 10:07:17.5757194	TACW2242222T1841	1	MeterValues	Meter (kWh): 0 | Charge (kW): 0 | SoC (%): -1	NULL
1287	2024-01-10 10:07:18.6238145	TACW2242222T1841	1	MeterValues	Meter (kWh): 0,003 | Charge (kW): 1,724 | SoC (%): -1	NULL
1288	2024-01-10 10:07:30.3202918	TACW2242222T1841	NULL	StopTransaction	Invalid	NULL
1289	2024-01-10 10:09:01.6103679	TACW2242222T1841	NULL		{"status": "Accepted"}	Unknown answer
1290	2024-01-10 10:09:03.4832857	TACW2242222T1841	1	StatusNotification	Info=null / Status=SuspendedEV / 	NoError
1291	2024-01-10 10:09:04.0258003	TACW2242222T1841	1	StatusNotification	Info=null / Status=Charging / 	NoError
1292	2024-01-10 10:09:04.6221071	TACW2242222T1841	1	StartTransaction	Invalid	NULL
1293	2024-01-10 10:09:05.1324052	TACW2242222T1841	1	StatusNotification	Info=null / Status=SuspendedEVSE / 	NoError
1294	2024-01-10 10:09:12.0507185	TACW2242222T1841	1	StatusNotification	Info=null / Status=Finishing / 	NoError
1295	2024-01-10 10:09:12.5903707	TACW2242222T1841	1	StatusNotification	Info=null / Status=Preparing / 	NoError
1296	2024-01-10 10:09:15.1538751	TACW2242222T1841	1	StartTransaction	Invalid	NULL