dallmann-consulting/OCPP.Core #45: The WebSocket received compressed frame when compression is not enabled.

DgnOnur, 2024-01-31:
Error Occurred in File: OCPPMiddleware.OCCP16.cs
 
Code Block: private async Task Receive16(ChargePointStatus chargePointStatus, HttpContext context)

Error-prone Code: WebSocketReceiveResult result = await chargePointStatus.WebSocket.ReceiveAsync(new ArraySegment<byte>(buffer),CancellationToken.None);

The error message is: "The WebSocket received compressed frame when compression is not enabled."

[image]

I connected three different devices to the same service. I am encountering an issue in one of the devices at this point. Can you help me?

dallmann-consulting, 2024-02-01:
That's weired. It is my understanding that websocket compression is disabled by default and must be enabled if "desired":
https://learn.microsoft.com/en-us/aspnet/core/fundamentals/websockets?view=aspnetcore-6.0#compression

_If compression of messages over WebSockets is desired, then the accept code **must** specify that it allows compression as follows..._

OCPP.Server does **NOT** enable compression this way. So it should be disabled!?
Maybe the affected charger ignores the handshake and sends the messages compressed anyway?

DgnOnur, 2024-03-05:
Hello, the problem was solved by trying the wss connection request.

closed 2024-03-05 by DgnOnur