dallmann-consulting/OCPP.Core #46: Add implementation for SetChargingProfile

pbonato, 2024-02-20:
Is possible to have the SetChargingProfile implemented ?
I need to change the MaxCurrent for the Wallbox so I can monitor the load and balance the power for charging the EV

dallmann-consulting, 2024-02-27:
I'm thinking about it. The limiting factor is not enough spare time.
The message itself is probably not enough. You also need an API to control the limit dynamically, right?
I thought of an MQTT interface for monitoring all activities and probably controlling the power limit.

andreasringdal, 2024-05-29:
@dallmann-consulting What are your thought on MQTT interface for this project. 
I am looking into modifying the OCPPMiddlweare to forward all incomming messages to IOT-Hub in order to enable individual billing for the charge points.

dallmann-consulting, 2024-08-27:
I have just added the SetChargingProfile and ClearChargingProfile messages. 
See the new Server-API.md for documentation.
Both messages are **NOT TESTED** yet!

closed 2024-09-17 by dallmann-consulting