dallmann-consulting/OCPP.Core #47: Transaction Authorized by RFID Tag and stopped with without RFID Tag!

dkozmaqi, 2024-03-04:
Hi,

Ive managed to run OCPP Core platform and connect two Siemens D Charger 300Kw DC.

I have an issue with uncompleted transaction that were authorized by RFID Tag and stopped by the car or power outage.

Uncompleted transactions in the photo below:
[image]
[image]

How can I modify the code to handle cases where the charging is stopped by the car/power outage and receive complete transaction data and unlock RFID Tag?

dallmann-consulting, 2024-03-06:
What version are you using?
There was a bug if the charger doesn't send an RFID tag with the stop transaction request. Our chargers are sending the tag from the start request if the cars forces a stop transaction. Possibly other chargers are doing the same. But the specification allows an empty RFID tag.
The behavior was fixed here:
https://github.com/dallmann-consulting/OCPP.Core/commit/6f27bfae2abf2452f4a372eab7ca8d793b2835b6

dkozmaqi, 2024-03-11:
I'm using the latest version v1.3.0!

For the moment I modified the configuration on appsettings.json = "DenyConcurrentTx": false, which is allowing me to continue use the same RFID tag.

I still receive uncompleted transactions.

dallmann-consulting, 2024-03-14:
When a transaction is stopped by the car, the charger still needs to send a StopTransaction message. The server just should not reject this message.
In case of a power failure, network outage etc. the chargers are supposed to store unsent messages and send/repeat them later. Our home chargers are doing that. We've had some server problems recently and all messages were transmitted later. I haven't seen any open transactions.

If your chargers are not doing this it will be difficult. You would need to assume stopped transactions from following status messages and meter values. Your chargers always seem to send a "0" starting meter value. You can't assume the stop value from the next beginning transaction.

dallmann-consulting, 2024-03-31:
Did you find the cause of your issues?

dkozmaqi, 2024-04-04:
This Siemens D Charger doesn't have DC meter. maybe the platform always reads the data from the meter.

I've used the same charger with commercial OCPP platform it always worked (except if we head power outage).

Maybe it's needed to make additional configuration on the OCPP server to be more compatible with Siemens D Charger!

Recently i connected ABB charger with DC meter its working fine, all the transaction are complete.

[image]
[image]
[image]

dallmann-consulting, 2024-04-04:
I'm not sure if I understand your comment correctly: 
_This Siemens D Charger doesn't have DC meter. maybe the platform always reads the data from the meter._

The OCPP server does not read anything from the chargers. They send messages to the server and that's all the information the server has. 
In your first screenshot all transactions start with a "0" value. If the charger send a "StopTransaction" message with a higher meter value all is fine. The problem starts when a StopTransaction message gets lost which seems to be the cause in your szenario.
The server could potentially assume a stopped transaction when an "available" status message is received. But then it still has no reliable end meter value. And in your case the next transaction starts with a 0 meter value. 
The chargers usually send meter values in certain intervals. The server would need to store these during transactions so that in case of an unfinished transaction, it can use the last transmitted value as the stop value. Did you mean that with your statement above?
Maybe the commercial platforms are doing that to avoid problems? But it is kind of difficult and to my understanding the chargers are supposed to store and resend messages until they are commited.