dallmann-consulting/OCPP.Core #49: Help Needed: Connecting Siemens VersiCharge AC Charger to Server on Same Wi-Fi Network

davidtomas98, 2024-04-05:
I need to connect a real Siemens charger, the VersiCharge AC. It's connected to the same Wi-Fi network as the server. Can you explain how I should do it?

davidtomas98, 2024-04-05:
I have connected the charger to the Wi-Fi network, configured a user tag, and created the charger within the server along with the ID of the tag that had been enabled for the charger. However, when I pass the tag and unlock the charger to charge, it does not show as connected on the web page

davidtomas98, 2024-04-05:
I have to configure something on the charger besides the Wi-Fi connection?

dallmann-consulting, 2024-04-05:
You need to configure the URL of your server in your charger. It is basically the same like with the simulator. It is not enough just to use the same wifi connection/network.
From googling it seems that Siemens doesn't allow you to configure that yourself:
https://monta.com/uk/help-center/siemens-versicharge-installation-guide/

They state that you need to contact the support and they reconfigure your charger - probably via their cloud connection.

davidtomas98, 2024-04-05:
You should send an email to support requesting the change to this address if the server is hosted locally: ws://localhost:9019/OCPP/{charger_id}. Once they confirm the change, I will need to create the charger and its tags within the database, similar to what I would do in the simulator. Finally, I will pass the admin tag to the charger to add user tags. When a user tag is passed, will it connect?

davidtomas98, 2024-04-05:
I would have to send an email to support requesting the change to this address if the server is hosted locally: ws://localhost:9019/OCPP/{charger_id}. Once they confirm the change, I will need to create the charger and its tags within the database, similar to what I would do in the simulator. Finally, I will assign the admin tag to the charger to add user tags. When a user tag is passed, will it connect?

davidtomas98, 2024-04-05:
And also, another thing please, I've noticed that the charger only accepts WSS connections. Would I need to change something in the code?

dallmann-consulting, 2024-04-05:
"localhost" means the same computer which would be the charger. That's wrong. Please try to setup the server and simulator on different machines, You need an IP address or DNS name in your network.
WS vs. WS**S** is the same like http and http**s**. Your server (OCPP.Core server) needs an SSL certificate for https connections. Please ask Siemens if this is really necessary. If you are working in a private network this seems not really necessary.

davidtomas98, 2024-04-17:
Now I've been told that they have set this up: ws://desktop-8f4o9rj:9019/OCPP and GROUPNAME: ALDROVE. Now, to test the functionality, I'll have to create the charger with its ID and put the ID of a tag, and theoretically it should work? I don't understand why they asked me for the GROUPNAME.

davidtomas98, 2024-04-17:
desktop-8f4o9rj" is the name of the local server where I have the database.

davidtomas98, 2024-04-17:
I've seen what the target group is. How do I specify the target group to OCPP on the charger, or is it unnecessary?

davidtomas98, 2024-04-17:
[image]

davidtomas98, 2024-04-17:
[image]

On the OCPP server, I have the following table of tags, and I've added two Siemens cards to the Aldrove group. However, when I swipe the card through the charger, it doesn't unlock to start the charging. What could be happening?

davidtomas98, 2024-04-17:
They have already changed my WebSocket address and they have set it to my OCPP server's address, where I have the database, which is this: ws://DESKTOP-8F4O9RJ:9019/OCPP. I have also added the charger with its ID to the database, as well as the tags, as I have shown you. However, I don't understand why it's not working.

dallmann-consulting, 2024-04-17:
Please check the message dump. All incoming and outgoing messages are written there as files (with default configuration). Are there any files? If not, the communication does not work. You can also check the log files on the server. You should see requests and processing information there.
If you see activity:  clear the dump directory and then present your card to the charger. You should see an authorization request and the answer from the server.

davidtomas98, 2024-04-17:
Please, I'm new to all this, could you be more specific? Where can I check the message dumping route? How could I check the log files on the server, and how do I delete the dump directory? I'm using SQL Server Management Studio 2022.

davidtomas98, 2024-04-17:
The charger I have is the Siemens VersiCharge AC model. To activate a card, you need to swipe the administrator card first, then the user card, and then again the administrator card. Do I need to do the same in this case if now the OCPP points to my server, and I have created the card within the database?

davidtomas98, 2024-04-18:
I've already checked the 'messagelog' table on the server, and there are no records of communication between the charger and the server. One point I need to mention is that I have both admin and user cards, and the charger manual instructs me to swipe the admin card to set up a user card. I've done this, so when I try the user card, it beeps but doesn't unlock, and there are no entries in the 'messagelog' table.

davidtomas98, 2024-04-18:
In the messagedump, I see this, but I don't know which charge point it is : [2,"ugXrsAnO10NtdcXFjHFZArBg81N9vacOsyUS","BootNotification",{"chargePointVendor":"AVT-Company","chargePointModel":"AVT-Express","chargePointSerialNumber":"avt.001.13.1","chargeBoxSerialNumber":"avt.001.13.1.01","firmwareVersion":"0.9.87","iccid":"","imsi":"","meterType":"AVT NQC-ACDC","meterSerialNumber":"avt.001.13.1.01"}]

davidtomas98, 2024-04-19:
I'm trying to run the simulator on another machine within the same network, but it doesn't interact at all. I'm typing this: ws://192.168.2.221:9019/OCPP/12 where the IP is that of the server, and I've already verified that it responds to ping and connects.

davidtomas98, 2024-04-19:
[image]
From the other machine, I check the connection to the server and see that it is open. However, when I enter the URL in the simulator and set it correctly, I click on 'Connect' and nothing happens.

chalonet, 2024-05-07:
If I manage to solve the issue, I will talk to the Siemens guys and tell them to use the WS URL and put the IP instead of the server name, then everything should work perfectly. Thank you very much for responding and for the help.

chalonet, 2024-05-07:
Yes, I managed to solve the problem. I spoke with the Siemens team and told them to change the WS URL and use the IP instead of the server name. Then everything worked perfectly. Thank you very much for responding and for the help.

closed 2024-05-08 by dallmann-consulting