dallmann-consulting/OCPP.Core #50: Improving performance and reliability

vadrsa, 2024-04-05:
Hi,
I maintain a library called [SimpleR](https://github.com/vadrsa/SimpleR). In a nutshell, it is a fork of SignalR without the custom protocols allowing you to write very fast pure WebSocket servers.

So you basically get all the benefits of SignalR, like advanced buffer management, low allocation, very good integration with asp.net core(routing, authentication, etc...) but still have the ability to integrate with custom protocols like OCPP.

I have started to write a helper package for OCPP, to make it simpler to write OCPP servers using SimpleR(pun intended) and would love to hear what you think can be beneficial to you. If you are interested in using such a library, let me know and I'll try to help as best I can.

JeremyMahieu, 2024-04-05:
This is an ad, not an issue. Whatever benefit it would give, it comes with a massive downside of relying on an unknown repo. This  doesn't use that much SignalR anyway and the performance benefits would be miniscule.

vadrsa, 2024-04-05:
What an unnecessarily rude reply. You are feee to close any issue on your super know repo👍

dallmann-consulting, 2024-04-08:
Cool down guys ;-)
I've had a very quick look at your project. It seems to be an interesting helper for other projects. But I don't see an advantage for OCPP.Core yet. Probably more than 99% of the code is related to the OCPP logic. And if I'm not mistaken SimpleR doesn't help with this, right?
As I said - I still find it very interesting and might use it in other projects.
Thanks for pointing me to this.

vadrsa, 2024-04-08:
> Cool down guys ;-)
>
> I've had a very quick look at your project. It seems to be an interesting helper for other projects. But I don't see an advantage for OCPP.Core yet. Probably more than 99% of the [...]

First of all, I'm sorry if the title of the issue came across as rude. I don't mean nor can I earn any money off of this project to advertise it. My motivation comes from the fact that I myself had to write an OCPP server for one of the projects I have worked on and didn't find any scalable alternative to SignalR, which can't be used in such scenarios.

Regarding the helpfulness of SimpleR, it can help with the transport layer, written in OCPPMiddleware. Using SimpleR will remove a lot of memory allocations resulting in the ability to maintain more connections. Also it will allow you to integrate more with aspnet core e.g. write the authentication logic using standard aspnet core constructs(AuthenticationHandler, ClaimsPrincipal) and use route parameters. Depending on your need to scale the number of connections, it might be worth it or not.

I am writing an OCPP protocol helper for SimpleR and one of my main questions here was if you had any suggestions as to what would be useful without making it too opinionated(not allow the request/response classes to be defined by the user as he likes, etc...) I am thinking just parsing to the level of OcppCall, OcppCallResult, OcppCallError will be beneficial.

I can do a simple PoC on top of OCPP.Core if that is ok with you.

Of course, I understand that using the lib is not the first choice because of reasons and that is not what I am pushing for. There aren't many OCPP implementations out there in .net and I want to understand the usefulness of my lib myself. So any feedback and suggestions are highly appreciated.