dallmann-consulting/OCPP.Core #54: No current status could be retrieved from the OCPP server

WELIZARY, 2024-04-23:
The charging stations work and are displayed correctly, but this error is always lit at the bottom

[image]

The following error appears in the server log
[image]

JeremyMahieu, 2024-04-23:
Probably the database queries are starting to take too long (longer than 4s) due to slow database views and lack of indexes. I fixed this in [this pull request](https://github.com/dallmann-consulting/OCPP.Core/pull/53), give it a try. 

You can use [this fork](https://github.com/JeremyMahieu/OCPP.Core) where a few of these pull requests are already merged with the main.

dallmann-consulting, 2024-04-23:
The Web-UI contacts the server to get the connection status of all charge points and connectors. The appsettings.json contains the URL and a secret key (see the matching key in the server settings ) for this status request:

```
  "ServerApiUrl": "http://localhost:8081/API",
  "ApiKey": "36029A5F-B736-4DA9-AE46-D66847C9062C",
```

If this request fails you get the above error message. 

Regarding the charging information shown:
Usual home chargers are only sending updated meter valued periodically. They don't send a SoC and the don't send power values. The Web-UI currently only shows the elapsed time, since the transaction startet (the tile is red). When the transaction has stopped (the tile is green) it shows the charged amount (kWh) of the last transaction. That is what you see in the screenshot.
If you are using the included modified chargepoint simulator you can also send a SoC and that gets displayed. But home chargers ususally don't send this information.
I already had the idea to also show the charged amount (until the latest transmitted meter value) in the active status (red tile) and the charging power could be calculated from the time between the last two meter values. But is this really interesting for a home charger with limited power? Most people just plug in their cars at home or work and the car charges itself to a certain determined level. It didn't seem very useful to me.

dallmann-consulting, 2024-04-23:
Yes, but if you look at the messages every minute you will probably see that they only contain updated meter values. This is currently not displayed (as stated above) because it doesn't seem very usefull with slow chargers. If I want to see the progress I check the SoC in my car and not the currently charged kWh.

closed 2024-04-23 by WELIZARY

dallmann-consulting, 2024-05-06:
Sorry, my request goes to JeremyMahieu.