dallmann-consulting/OCPP.Core #56: [Feature] Charge report

JeremyMahieu, 2024-05-21:
This is a feature request but I would be inclinded to make this feature myself. Opening an issue first so it can be discussed first, avoid double work.

1) Make an extra page in the menu Administration > Charge report
2) Have a date time span selector to indicate begin time, end time
3) Two tables with charge data: One that says the amount of energy charged per person "123kWh". Grouped by the "group" of a chargetag. And another table that gives details of the person with each charge session. This table only appears if you click a person.
4) 2 exports to csv/xlsx (using ClosedXML library), one with the aggregated data as in the first table, and another one that gives all charge transactions in the selected timespan

Here is a rough example, although I would do the layout totaly different so it matches the existing layout.
[image]

JeremyMahieu, 2024-05-24:
Can discuss in the pull request

closed 2024-05-24 by JeremyMahieu

dallmann-consulting, 2024-08-07:
I had to redesign some stuff. The transactions are decoupled from the charge tags by design. The charge tags are meant like an optional part (left join). Your queries and the integration of the "ChargeTag" model in "Transaction" didn't work in such cases because of the inner join logic.