dallmann-consulting/OCPP.Core #63: [Feature] Exporting (prometheus) metrics

JeremyMahieu, 2024-06-26:
It would be nice if this OCPP server had an endpoint /metrics where the actual values for each chargepoint are written. Then we could use something like influxdb or premetheus/grafana to make a dashboard of the chargeing infrastructure.

nuget has prometheus client libraries that produce this file, just need to set the values

Some example data of what a metrics file could look like (just some invented metrics I don't know what would be usefull,
```
# HELP ocpp_chargepoint_sessions_total Total number of charging sessions
# TYPE ocpp_chargepoint_sessions_total counter
ocpp_chargepoint_sessions_total{chargepoint="CP1", rfid_tag="TAG123"} 120
ocpp_chargepoint_sessions_total{chargepoint="CP2", rfid_tag="TAG456"} 85
ocpp_chargepoint_sessions_total{chargepoint="CP3", rfid_tag="TAG789"} 98

# HELP ocpp_chargepoint_energy_consumed_kwh Total energy consumed by the chargepoint in kilowatt hours
# TYPE ocpp_chargepoint_energy_consumed_kwh counter
ocpp_chargepoint_energy_consumed_kwh{chargepoint="CP1", rfid_tag="TAG123"} 24.005
ocpp_chargepoint_energy_consumed_kwh{chargepoint="CP2", rfid_tag="TAG456"} 19.877
ocpp_chargepoint_energy_consumed_kwh{chargepoint="CP3", rfid_tag="TAG789"} 21.532

# HELP ocpp_chargepoint_uptime_percentage Percentage of uptime for the chargepoint
# TYPE ocpp_chargepoint_uptime_percentage gauge
ocpp_chargepoint_uptime_percentage{chargepoint="CP1", rfid_tag="TAG123"} 99.8
ocpp_chargepoint_uptime_percentage{chargepoint="CP2", rfid_tag="TAG456"} 98.5
ocpp_chargepoint_uptime_percentage{chargepoint="CP3", rfid_tag="TAG789"} 99.1

# HELP ocpp_chargepoint_current_sessions_active Number of currently active charging sessions
# TYPE ocpp_chargepoint_current_sessions_active gauge
ocpp_chargepoint_current_sessions_active{chargepoint="CP1", rfid_tag="TAG123"} 1
ocpp_chargepoint_current_sessions_active{chargepoint="CP2", rfid_tag="TAG456"} 2
ocpp_chargepoint_current_sessions_active{chargepoint="CP3", rfid_tag="TAG789"} 1

# HELP ocpp_chargepoint_error_count_total Total number of errors encountered by the chargepoint
# TYPE ocpp_chargepoint_error_count_total counter
ocpp_chargepoint_error_count_total{chargepoint="CP1", rfid_tag="TAG123"} 5
ocpp_chargepoint_error_count_total{chargepoint="CP2", rfid_tag="TAG456"} 12
ocpp_chargepoint_error_count_total{chargepoint="CP3", rfid_tag="TAG789"} 8

# HELP ocpp_chargepoint_current_charge_current_amp Current charge current in amperes
# TYPE ocpp_chargepoint_current_charge_current_amp gauge
ocpp_chargepoint_current_charge_current_amp{chargepoint="CP1", rfid_tag="TAG123"} 16.5
ocpp_chargepoint_current_charge_current_amp{chargepoint="CP2", rfid_tag="TAG456"} 20.4
ocpp_chargepoint_current_charge_current_amp{chargepoint="CP3", rfid_tag="TAG789"} 32.0

# HELP ocpp_chargepoint_charge_limit_amp Charge limit in amperes
# TYPE ocpp_chargepoint_charge_limit_amp gauge
ocpp_chargepoint_charge_limit_amp{chargepoint="CP1", rfid_tag="TAG123"} 32
ocpp_chargepoint_charge_limit_amp{chargepoint="CP2", rfid_tag="TAG456"} 32
ocpp_chargepoint_charge_limit_amp{chargepoint="CP3", rfid_tag="TAG789"} 32
```

dallmann-consulting, 2024-06-27:
I thought about using MQTT for this. And this way an external system (PV controller) could also set/limit the available power.

JeremyMahieu, 2024-06-28:
That could work.

dallmann-consulting, 2024-07-25:
I'm thinking about this one as well. If I'm not mistaken, the server needs an endpoint "/metrics" which could be implemented in the existing middlware (like the API) oder start a parallel kestrel web server for this. Both solutions have their pros and cons.

What usefull metrics would you expect?

[Gauge]Chargers online (like 3 of 5)
[Summery]Connectors status (free/in Use)

[Histogram] Errors (like bad messages from chargers)
[Histogram] RFID auth success/errors

Data about running transactions seems difficult. I haven't seen any AC charger (no HPC) that does send electric values (like AMPs etc.). The server can only calculate the power (kW) from meter-messages and the interval between them.

What values would you expect?