dallmann-consulting/OCPP.Core #64: Problems to Login

DavidGG2001, 2024-07-17:
Hi,
I've build the program and when I'm about to login  I can't.
[image]

mstrini, 2024-07-17:
And you have used the default-credentials from https://github.com/dallmann-consulting/OCPP.Core/blob/main/OCPP.Core.Management/appsettings.json ? That has worked for me last time...

DavidGG2001, 2024-07-18:
> And you have used the default-credentials from https://github.com/dallmann-consulting/OCPP.Core/blob/main/OCPP.Core.Management/appsettings.json ? That has worked for me last time...

Yes I prove it but doesn't work. Could it be that I created my OCPP.Core database wrong?

DavidGG2001, 2024-07-18:
[image]

The error is that can't connect correctly with the database

JeremyMahieu, 2024-07-18:
You are using a empty connection string, make sure you didn't comment it out in the appsettings.

DavidGG2001, 2024-07-18:
I tried with the SQLite and the SQLServer and the 2 options are not working

dallmann-consulting, 2024-07-18:
Please try to configure SQLite and send us the logs here.

DavidGG2001, 2024-07-18:
I tried with SQLite and now I have this problem
[image]
 The logs of the manager are this:
: Microsoft.AspNetCore.Server.Kestrel[0] @ 2024-07-18T09:37:42.1837320+02:00
      Overriding address(es) 'http://192.168.2.213:8082'. Binding to endpoints defined via IConfiguration and/or UseKestrel() instead.
info: OCPP.Core.Management.Controllers.AccountController[0] @ 2024-07-18T09:38:09.3823888+02:00
      User 'admin' logged in
trce: OCPP.Core.Management.Controllers.HomeController[0] @ 2024-07-18T09:38:09.4060899+02:00
      Index: Loading charge points with latest transactions...
fail: OCPP.Core.Management.Controllers.HomeController[0] @ 2024-07-18T09:38:11.5567311+02:00
      Index: Error in status web request => No se puede establecer una conexión ya que el equipo de destino denegó expresamente dicha conexión. (192.168.2.213:8081)
System.Net.Http.HttpRequestException: No se puede establecer una conexión ya que el equipo de destino denegó expresamente dicha conexión. (192.168.2.213:8081)
 ---> System.Net.Sockets.SocketException (10061): No se puede establecer una conexión ya que el equipo de destino denegó expresamente dicha conexión.
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError error, CancellationToken cancellationToken)
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
   at System.Net.Sockets.Socket.<ConnectAsync>g__WaitForConnectWithCancellation|285_0(AwaitableSocketAsyncEventArgs saea, ValueTask connectTask, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.ConnectAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.CreateHttp11ConnectionAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.AddHttp11ConnectionAsync(QueueItem queueItem)
   at System.Threading.Tasks.TaskCompletionSourceWithCancellation`1.WaitWithCancellationAsync(CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.SendWithVersionDetectionAndRetryAsync(HttpRequestMessage request, Boolean async, Boolean doRequestAuth, CancellationToken cancellationToken)
   at System.Net.Http.DiagnosticsHandler.SendAsyncCore(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.RedirectHandler.SendAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpClient.<SendAsync>g__Core|83_0(HttpRequestMessage request, HttpCompletionOption completionOption, CancellationTokenSource cts, Boolean disposeCts, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)
   at OCPP.Core.Management.Controllers.HomeController.Index() in C:\Users\Proacit\Desktop\OCPP.Core-main\OCPP.Core.Management\Controllers\HomeController.Index.cs:line 91
fail: Microsoft.EntityFrameworkCore.Database.Command[Microsoft.EntityFrameworkCore.Database.Command.CommandError] @ 2024-07-18T09:38:12.4897093+02:00
      Failed executing DbCommand (16ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
      SELECT "c"."ChargePointId", "c"."ConnectorId", "c"."ConnectorName", "c"."LastStatus", "c"."LastStatusTime", "c"."LastMeter", "c"."LastMeterTime", "t"."TransactionId", "t"."StartTagId", "t"."StartTime", "t"."MeterStart", "t"."StartResult", "t"."StopTagId", "t"."StopTime", "t"."MeterStop", "t"."StopReason"
      FROM "ConnectorStatus" AS "c"
      LEFT JOIN "Transactions" AS "t" ON "c"."ChargePointId" = "t"."ChargePointId" AND "c"."ConnectorId" = "t"."ConnectorId"
      WHERE "t"."TransactionId" IS NULL OR "t"."TransactionId" = (
          SELECT MAX("t0"."TransactionId")
          FROM "Transactions" AS "t0"
          WHERE "t0"."ChargePointId" = "c"."ChargePointId" AND "t0"."ConnectorId" = "c"."ConnectorId")
fail: Microsoft.EntityFrameworkCore.Query[Microsoft.EntityFrameworkCore.Query.QueryIterationFailed] @ 2024-07-18T09:38:12.5051967+02:00
      An exception occurred while iterating over the results of a query for context type 'OCPP.Core.Database.OCPPCoreContext'.
      Microsoft.Data.Sqlite.SqliteException (0x80004005): SQLite Error 1: 'no such table: ConnectorStatus'.
         at Microsoft.Data.Sqlite.SqliteException.ThrowExceptionForRC(Int32 rc, sqlite3 db)
         at Microsoft.Data.Sqlite.SqliteCommand.PrepareAndEnumerateStatements()+MoveNext()
         at Microsoft.Data.Sqlite.SqliteCommand.GetStatements()+MoveNext()
         at Microsoft.Data.Sqlite.SqliteDataReader.NextResult()
         at Microsoft.Data.Sqlite.SqliteCommand.ExecuteReader(CommandBehavior behavior)
         at Microsoft.Data.Sqlite.SqliteCommand.ExecuteDbDataReader(CommandBehavior behavior)
         at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteReader(RelationalCommandParameterObject parameterObject)
         at Microsoft.EntityFrameworkCore.Query.Internal.SingleQueryingEnumerable`1.Enumerator.InitializeReader(Enumerator enumerator)
[... cut ...]
   at Microsoft.EntityFrameworkCore.Storage.NonRetryingExecutionStrategy.Execute[TState,TResult](TState state, Func`3 operation, Func`3 verifySucceeded)
   at Microsoft.EntityFrameworkCore.Query.Internal.SingleQueryingEnumerable`1.Enumerator.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Linq.Enumerable.ToList[TSource](IEnumerable`1 source)
   at OCPP.Core.Management.Controllers.HomeController.Index() in C:\Users\Proacit\Desktop\OCPP.Core-main\OCPP.Core.Management\Controllers\HomeController.Index.cs:line 140

dallmann-consulting, 2024-07-18:
Are you sure that you are using the provided sqlite-file? This file contains the missing table:

[image]

I would suggest you delete the current sqlite-file and download it again from this repoitory.

DavidGG2001, 2024-07-18:
I think that I finally found the problem
[image]
To solve it I have only to put local host in root certificates or something else?
Thanks

JeremyMahieu, 2024-07-18:
You can put `;TrustServerCertificate=True` in the connection string. Not sure why the repo has =false, but this is a very common one. If it's false you don't need to mention it. Most people just set it to be True if it's a local server, no need to go over the top with certificates if it's localhost anyway.

DavidGG2001, 2024-07-19:
I tried once more and it worked without any changes. 
Thanks

DavidGG2001, 2024-07-19:
Very grateful for you help :)

closed 2024-07-19 by DavidGG2001