dallmann-consulting/OCPP.Core #65: Question about Charger Status and Remote Transactions

DavidGG2001, 2024-07-19:
I've connected a Siemens charger, my questions is if now the program without any change can notify me the status of the charger or I've to do any implementation?
In the other hand, I want to start remote transactions and stop it when the the timer of the user is 0. Anyone have an idea of how to do it ?I've try it modifying the OCPP Middleware and OCPP controller but I haven't achieve it.

Thanks for your help :)

dallmann-consulting, 2024-07-19:
Can the system notify you, when a status change happens?
Currently not in a passive way. The server keeps the state of all chargers/connectors in memory and also writes it to the database (table ConnectorStatus). You have 2 options:
1.) Read it periodically from the database.
2.) Read it periodically from the server via the API (also polling). The Web-UI does the same to show the status of all connectors.
I'm plannung to add an MQTT Client to push status values to a server. That way the MQTT server could also set charging limits - like what your PV system currently provides.

Start and stop transactions remotely.
These messages are not implemented in this version. This version is intended for home use and starting/stopping remote transations very often is a commercial scenario.

DavidGG2001, 2024-07-23:
I'm trying to implement start remote because I want to start the transaction through the management page and stop it when I want or when my timer is over. I have created an ControllerOCPP16.RemoteStartTransaction but I don't know if I'm going on the correct way.

Finally the other issue I tried reading periodically from the database and it worked, thanks :)

closed 2024-09-26 by dallmann-consulting