dallmann-consulting/OCPP.Core #66: [Feature] Enhanced User Management and Functions for OCPP.Core

TSK7M, 2024-07-29:
### Feature Request: Enhanced User Management and Functions for OCPP.Core

#### Summary
I am currently using OCPP.Core on a Windows 2022 Server with IIS and have several suggestions for improving user management and adding additional features. These enhancements aim to improve user experience and control over the charging infrastructure.

#### Requirements

1. **RFID Tag Assignment and User Management**
    - Ability to assign RFID tags to users who can log into the web interface.
    - Users can only view the charging status of their own RFID tags.
    - Other charging stations are displayed only as "available" or "occupied".
    - Option to hide specific stations for individual users.

2. **Permission Management**
    - Assign RFID tags to specific charging stations, allowing users to charge only at the stations they are permitted to use.
    - Management of permissions per charging station.

3. **Reservation System**
    - Ability to reserve charging stations for specific time periods.
    - Management and display of reservations in the web interface.

4. **Charging Limits and Budgets**
    - Setting charging limits/budgets for RFID tags, e.g., daily, weekly, or monthly (e.g., max. 25 kWh).
    - Management of charging limits for users with multiple RFID tags.

5. **Deletion Functions**
    - Ability to delete RFID tags, charging stations, and other relevant data.

6. **Export Functions**
    - Extend the existing CSV export to include a PDF export function.
    - Apply filters in the web interface before exporting data.

#### Summary
These enhancements would significantly improve user management and control over the charging infrastructure. Particularly helpful would be the ability to view charging statuses only for one's own RFID tags, assign permissions to specific charging stations, implement a reservation system, set charging limits, and improve export functionality.

I am very satisfied with the current solution and look forward to future enhancements.

NRGkick + KEBA

[image]

[image]

[image]

TSK7M, 2024-07-29:
### Benefits and Impact of Proposed Features

**Enhanced User Experience**

- **Personalized Views and Permissions**: By providing personalized views and permissions, the user interface becomes more intuitive and organized. Each user sees only the information relevant to them, such as the status of their own RFID tags and reserved charging stations. This reduces information overload and simplifies navigation.
- **Individual Account Access and Reservation System**: Users have individual accounts that allow them to log into the web interface. Through their account, they can access, monitor, and export their specific data as needed. Additionally, they can book charging stations using an integrated reservation system. A calendar visually displays which time slots are booked or available, enabling efficient planning and management of their charging times.

**Increased Control and Security**

- **Permission Management**: Detailed permission management ensures that only authorized users can access specific charging stations. This prevents unauthorized access and misuse of the charging infrastructure. Each charging station can be assigned to specific RFID tags, so only authorized users can use the station.
- **Charging Limits for Cost Control**: Implementing charging limits (e.g., daily, weekly, or monthly) for RFID tags helps monitor energy consumption and control costs. This ensures that usage stays within set budgets, and users can plan and adjust their charging activities to stay within the prescribed limits.

**Operational Efficiency**

- **Reservation System and Calendar**: An integrated reservation system allows users to book charging stations for specific time periods in advance. A visual calendar clearly shows booked and available time slots, allowing users to plan their charging times efficiently. This ensures optimal use of the charging infrastructure and minimizes conflicts in station usage.
- **Optimization of Station Availability**: Targeted reservation and scheduling help avoid bottlenecks and optimize station utilization. This reduces wait times and increases overall user satisfaction by ensuring that charging stations are available when needed.

**Enhanced Reporting**

- **Improved Export Functions**: The ability to export data in both CSV and PDF formats significantly enhances reporting capabilities. Users can export their charging activities and other relevant data in the format that best suits their needs.
- **Filtering Before Export**: Users can apply specific filters before exporting data to sort and refine it according to certain criteria. This allows for targeted and detailed reports tailored to the needs of users or management. Enhanced reports support more precise data analysis and help in making informed decisions.
- **Better Data Analysis**: Combining filtered data with flexible export options provides deeper insights. Companies can identify trends and patterns, plan charging infrastructure more efficiently, and make strategic decisions based on detailed analyses.

These features provide comprehensive improvements in user experience, control, efficiency, and data analysis, significantly enhancing the management and utilization of the charging infrastructure.

TSK7M, 2024-07-31:
Hi 👋

I wanted to ask for your help with some new features and improvements for the OCPP.Core project. These enhancements aim to improve user management and overall functionality.

Here’s a quick summary of the requested features:

1. **User Management Enhancements:**
   - Assign RFID tags to users.
   - Users can only view their own RFID tag statuses.
   - Display other stations as "available" or "occupied" for user.
   - Hide specific stations for individual users.

2. **Permission Management:**
   - Assign RFID tags to specific stations.
   - Manage permissions per station.

3. **Reservation System:**
   - Reserve stations for specific times.
   - Display reservations in the web interface.

4. **Charging Limits and Budgets:**
   - Set daily, weekly, or monthly charging limits.
   - Manage limits for users with multiple RFID tags.

5. **Deletion Functions:**
   - Delete RFID tags, stations, and relevant data.

6. **Export Functions:**
   - Extend CSV export to include PDF.
   - Apply filters before exporting data.

These features would greatly improve user experience and control over the charging infrastructure. 

Thanks a lot for your help!

TSK7M, 2024-07-31:
### Overview of All Connected Charging Stations

**Functions:**
- User/Tag/RFID functionality for station and admin
- Charging station status
- Restart charging station (Admin only)
- Disconnect from charging station (Admin only)
- Start and stop charging process - Web User get a Tag / virtual RFID

### Report

- Display charging sessions per user in user accounts
- Filter charging sessions by RFID card, time period, and price per kWh
- Export reports as PDF or CSV files

**Information in the Report:**
- Used RFID card
- Start/end date and time
- Name of the charging station
- Duration
- Charged electricity
- Price
- Total period, number of charging sessions, total charged electricity, total charging duration, price per kWh, total cost
- Set company name for reports

### RFID Cards

- Manage RFID cards

**Functions:**
- Add RFID card (name, serial number, expiration date)
- Edit RFID card (name, expiration date, status, currency profile)
- Export/import RFID cards (CSV format)

### Configuration

- Set currency and cost per kWh and profiles for tags (RFID)
- Change application language (English, German)

TSK7M, 2024-07-31:
### Feature Request: Extended Features for Charging Rates and Cost Export

Dear OCPP.Core Development Team,

I would like to propose a set of features that would significantly enhance the user management and cost tracking functionalities of the OCPP.Core system. These enhancements are aimed at improving the flexibility and accuracy of charging rate management, cost calculation, and export capabilities. Here are the detailed feature requests:

#### Charging Rates per Charging Station/Wallbox

**Defining Charging Rates**

- **Implement functionality to define different charging rates per charging station/wallbox.**
- **Create an administrative interface that allows defining and managing rates (e.g., based on time, day, or consumption).**

**Integration of Charging Rates into the System**

- **Adjust the backend logic to account for the defined rates during the use of charging stations.**
- **Ensure that the rates can be dynamically adjusted and correctly applied when using the stations.**

#### Cost Calculation and Display

**Calculating Costs per User**

- **Implement a function to calculate costs based on the defined rates and usage of each charging station/wallbox.**
- **Ensure that costs are calculated in real-time or at regular intervals for accurate billing.**

**Displaying Costs in the Web Interface**

- **Integrate a display that shows users their costs based on usage and the defined rates.**
- **Ensure that costs are displayed clearly and understandably.**

#### Extension of Export Functions

**Exporting Costs per User in CSV and PDF**

- **Extend the existing export function to include costs per user and day in CSV and PDF exports.**
- **Implement filters that allow sorting and adjusting data before export (e.g., by period, user, or charging station).**

**Automatic Billing**

- **Implement a function that automatically generates bills and exports them in CSV or PDF format.**
- **Ensure that the bills contain all necessary information for direct processing.**

### Flexible Charging Rates per User and Day/Time

**Defining Flexible Charging Rates**

- **Allow charging rates to be assigned per user, per day/time, and per charging station/wallbox.**
- **Develop a flexible rate management system where administrators can assign rates based on various combinations:**
  - **Global rates for all users and all stations.**
  - **User-specific rates for all their tags/stations.**
  - **Day-specific rates for all users.**
  - **Combination rates (e.g., specific user on a specific day at a specific station).**

**Integration of Flexible Rates**

- **Ensure the backend logic can handle and apply these flexible rate combinations during charging sessions.**
- **Update the administrative interface to support the detailed configuration of these rates.**

I believe these enhancements would greatly improve the user experience and administrative control within the OCPP.Core system. 

Thank you

TSK7M, 2024-08-01:
These adjustments represent a significant improvement in the user interface and functionality by enabling intuitive and efficient management of charging reports. The enhancements facilitate daily usage and provide better visibility of the energy consumption of various RFID tags. However, it would be extremely desirable if the remaining suggested modifications could also be implemented to further enhance the user experience and functionalities.

Thank you very much!

[image]

TSK7M, 2024-08-03:
NRGkick GPS Data Integration for Tariff and Charging Location Restrictions

The NRGkick device sends GPS data via OCPP, allowing for local tariff and charging availability restrictions. When an NRGkick device is connected using a specific TAG/RFID, it would be highly beneficial to link the GPS position with the RFID tag. This would enable location-based restrictions, ensuring that users can only charge at designated locations and apply tariffs based on their GPS positions. This feature is crucial for mobile charging, as it allows for the enforcement of location-specific charging policies and tariff management.

KETRF, 2025-01-29:
Hello everyone

Is there already an update on the improvements suggested in this issue?
Are there any plans for an implementation or roadmap for these features?

Thank you in advance for your feedback!

JeremyMahieu, 2025-01-29:
Depends when you will do a pull request I suppose.