dallmann-consulting/OCPP.Core #67: [BUG] OCPP.Core - main // System.ArgumentException: 'DbSet<ChargeTag>

KETRF, 2024-08-05:
The new source code unfortunately has an error. When I use the application after compiling, I get the following error when the transaction is stopped.

The issue does not occur with the version "OCPP.Core.V1.3.1.Portable.Framework.Dependent.zip".

1x Windows Server Std. 2022, IIS
1x Windows Server Std. 2022, MS-SQL DB 

The setup includes a separate SQL instance with sysadmin permissions for the IIS user and OCPP Core.

**ControllerOCPP16.StopTransaction.cs - Error:**

info: OCPP.Core.Server.ControllerOCPP16[0]
      UpdateConnectorStatus => Save ConnectorStatus: ID=XXXXXXX / Connector=1 / Status=(null) / Meter=1279.09
fail: OCPP.Core.Server.ControllerOCPP16[0]
      StopTransaction => Exception writing transaction: chargepoint=XXXXXXX / tag=XXXXXXX
      System.ArgumentException: The key value at position 0 of the call to 'DbSet<ChargeTag>.Find' was of type 'ChargeTag', which does not match the property type of 'string'.
         at Microsoft.EntityFrameworkCore.Internal.EntityFinder`1.FindTracked(Object[] keyValues)
         at Microsoft.EntityFrameworkCore.Internal.EntityFinder`1.Find(Object[] keyValues)
         at Microsoft.EntityFrameworkCore.Internal.InternalDbSet`1.Find(Object[] keyValues)
         at Microsoft.EntityFrameworkCore.DbContext.Find[TEntity](Object[] keyValues)
         at OCPP.Core.Server.ControllerOCPP16.HandleStopTransaction(OCPPMessage msgIn, OCPPMessage msgOut) in D:\Temp\OCPP.Core-main\OCPP.Core.Server\ControllerOCPP16.StopTransaction.cs:line 112

Can someone please assist me?

THX

dallmann-consulting, 2024-08-06:
Thanks for the hint. It is caused by the latest changes (charge report). I'm looking into it.

KETRF, 2024-08-06:
I thought so, but I'm really looking forward to the feature :) Thanks a lot!

KETRF, 2024-08-07:
[image]

I have now installed the software.

2024-08-07 18:16:21.613: info: Microsoft.EntityFrameworkCore.Database.Command[20101]
2024-08-07 18:16:21.613:       Executed DbCommand (13ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
2024-08-07 18:16:21.613:       SELECT 1
2024-08-07 18:16:21.628: info: Microsoft.EntityFrameworkCore.Database.Command[20101]
2024-08-07 18:16:21.628:       Executed DbCommand (8ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
2024-08-07 18:16:21.628:       SELECT OBJECT_ID(N'[__EFMigrationsHistory]');
2024-08-07 18:16:21.628: info: Microsoft.EntityFrameworkCore.Database.Command[20101]
2024-08-07 18:16:21.628:       Executed DbCommand (1ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
2024-08-07 18:16:21.628:       SELECT 1
2024-08-07 18:16:21.675: info: Microsoft.EntityFrameworkCore.Database.Command[20101]
2024-08-07 18:16:21.675:       Executed DbCommand (5ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
2024-08-07 18:16:21.675:       CREATE TABLE [__EFMigrationsHistory] (
2024-08-07 18:16:21.675:           [MigrationId] nvarchar(150) NOT NULL,
2024-08-07 18:16:21.675:           [ProductVersion] nvarchar(32) NOT NULL,
2024-08-07 18:16:21.675:           CONSTRAINT [PK___EFMigrationsHistory] PRIMARY KEY ([MigrationId])
2024-08-07 18:16:21.675:       );
2024-08-07 18:16:21.675: info: Microsoft.EntityFrameworkCore.Database.Command[20101]
2024-08-07 18:16:21.675:       Executed DbCommand (1ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
2024-08-07 18:16:21.675:       SELECT 1
2024-08-07 18:16:21.675: info: Microsoft.EntityFrameworkCore.Database.Command[20101]
2024-08-07 18:16:21.675:       Executed DbCommand (1ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
2024-08-07 18:16:21.675:       SELECT OBJECT_ID(N'[__EFMigrationsHistory]');
2024-08-07 18:16:21.691: info: Microsoft.EntityFrameworkCore.Database.Command[20101]
2024-08-07 18:16:21.691:       Executed DbCommand (1ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
2024-08-07 18:16:21.691:       SELECT [MigrationId], [ProductVersion]
2024-08-07 18:16:21.691:       FROM [__EFMigrationsHistory]
2024-08-07 18:16:21.691:       ORDER BY [MigrationId];
2024-08-07 18:16:21.691: info: Microsoft.EntityFrameworkCore.Migrations[20402]
2024-08-07 18:16:21.691:       Applying migration '20240405204318_TransactionsIndex'.
2024-08-07 18:16:21.738: info: Microsoft.EntityFrameworkCore.Database.Command[20101]
2024-08-07 18:16:21.738:       Executed DbCommand (16ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
2024-08-07 18:16:21.738:       CREATE INDEX [IX_Transactions_ChargePointId_ConnectorId] ON [Transactions] ([ChargePointId], [ConnectorId]);
2024-08-07 18:16:21.738: info: Microsoft.EntityFrameworkCore.Database.Command[20101]
2024-08-07 18:16:21.738:       Executed DbCommand (2ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
2024-08-07 18:16:21.738:       INSERT INTO [__EFMigrationsHistory] ([MigrationId], [ProductVersion])
2024-08-07 18:16:21.738:       VALUES (N'20240405204318_TransactionsIndex', N'8.0.3');

But I still need to run the test.

Thank you and best regards.

KETRF, 2024-08-07:
[image]

Starting and stopping with the app and tag worked without any issues.

Thank you very much!

closed 2024-08-07 by KETRF