dallmann-consulting/OCPP.Core #69: Question about load balancing

7WKTR, 2024-09-10:
Hi, I Just found Your project and I really think it's awesome!

I didn't find information about any load balancing - do You plan it or is it already included?
The easiest one would be to set some max allowed power value for building and compare it with value from some energy meter, for example Shelly 3EM via MQTT. Then send lower current value to all chargers.

Sorry if I missed a point of this project, maybe You have another solution to limit power of all chargers together? I just think, that in real world this functionality is necessary in facilities with more chargers.

JeremyMahieu, 2024-09-10:
This can be separate from OCPP, one has nothing to do with the other. We use node-red to collect grid connection meter data and set the current limit of the charging points via modbus or mqtt.

dallmann-consulting, 2024-09-11:
I think the answer is bit more complex.
The best and most reliable solution is always within the chargers itself. Our Keba chargers support an integrated load management. One charger is the master and all others are slaves. The master can also read values from an energy meter. The mash of chargers know the real energy consumption and can adapt dynamically. The chargers also have a failsafe mode - so when the communication between the chargers fail they will drop to a safe level to avoid overload.

OCPP doesn't have such mechanisms. And with OCPP the central server doesn't get constant energy values. The chargers are only sending meter values in certains intervals. 

I have just added the SetChargingProfile message to the server. That's the way to send a power limit to a charger. But that's far too slow and unreliable for a realtime load management.
I'm thinking of a semi dynamic scenario: Assume you have a safe limit of 16kW and 3 chargers. When the first car connects it is allowed the maximum (probably 11kW). When a second car connects both charger need to be set to 8kW. With three cars only 5kW.
It doesn't matter if and how fast a car really charges. The OCPP server doesn't know that. It can only assume that the car might always charge with the max. allowed power.
But even this scenario needs to be reliable. And this can be complex. The charger can't be reachable. Or they the reject the message. Or... 

Does this make sense?

7WKTR, 2024-09-11:
I am currently using my DIY version of the OpenEVSE charger and Home Assistant as the OCPP server.

I can read power values in "real-time." But "real-time" might mean something different to You. I know that operating at 20 kW when the allowed limit is 16 kW for 30 seconds is unlikely to be dangerous and won’t trip the circuit breaker.

However, you definitely have a point! The system should be safe, and I hadn’t considered short-term overcurrent situations. It turns out that load balancing isn’t as straightforward as I initially thought. My idea was efficient, but yours is safe. There should be a balance between the two.

I need to think about this for a while. ;)

dallmann-consulting, 2024-09-12:
When you have got your own little infrastructure you can possibly build the load management yourself.
As stated above the OCPP server now supports the SetChargingProfile message.
See the server API [here](https://github.com/dallmann-consulting/OCPP.Core/blob/main/Server-API.md)

Read the values from your power meter, calculate your allowed charging power and call
   `http://<server:port>/API/SetChargingLimit/mycharger/1/2000W`
for every charger and/or connector.

But always check the response. It must be "accepted"!

closed 2024-09-26 by dallmann-consulting