dallmann-consulting/OCPP.Core #71: Cannot find the object "Transactions"

SayedaliSayedb, 2024-09-21:
Hello Dear
I have problem to run this project 
This error massage occurs during migration
Thank you for guiding me

Microsoft.Data.SqlClient.SqlException: 'Cannot find the object "Transactions" because it does not exist or you do not have permissions.

[image]

dallmann-consulting, 2024-09-21:
Is it possible that you have an empty database?
The server doesn't create the tables etc, itself. See the Script "Create_OCPP.Core.sql" in the "SQL-Server" directory.
Also make sure that the connection string points to the correct database.

JeremyMahieu, 2024-09-21:
This is why the first migration should be to create the database but it's like this historically.

SayedaliSayedb, 2024-09-21:
Thank you for your reply

I have tested this project on my computer.

Do I need to migration or do something before the implementation of the project?
Is it possible to guide step by step what should be done before the implementation of the project?

I followed the previous error like this, but it did not open
After running, I saw that the database was created but there was no table called Transactions, I created it myself in the database and put two values ​​"ChargePointId", "ConnectorId" and the ocpp.core.server project was executed and did not give an error.
But the ocpp.core.managment project is facing an error

[image]
[image]

JeremyMahieu, 2024-09-21:
Don't create it yourself. That is just one table of many. Upon first startup it should have run "Create_OCPP.Core.sql" which it seems like it didn't.

SayedaliSayedb, 2024-09-21:
Yes, you are right

I completely deleted the database and rebuilt it, then I ran Create_OCPP.Core.sql and it created the tables, then I ran the project and this problem has occurred.

[image]
[image]

JeremyMahieu, 2024-09-21:
Seems like you deleted the migration history table but not the transactions table, so it tried to create the index twice. Or it's something else.

SayedaliSayedb, 2024-09-21:
I acted like this
1- I deleted the previous database OCPP.Core
2- I created a new OCPP.Core database
3- I ran the Create_OCPP.Core.sql file and saw that it created the tables
4- I ran the ocpp.core.server project
 I encountered this error

Do I need to do anything else?

JeremyMahieu, 2024-09-21:
It seems that index is in both the migration and the initial sql file. This is not ok, not your fault.

You can temporarily fix it by editing the migration history table yourself and add "20240405204318_TransactionsIndex" version "8.0.3". Then it won't try to do the migration again and start up propertly.

SayedaliSayedb, 2024-09-21:
I didn't understand what you meant
Is it possible to guide me on what changes should I make in the file "20240405204318_TransactionsIndex"?

SayedaliSayedb, 2024-09-21:
Thanks for your tips
The OCPP.Core.Server project was executed
[image]

closed 2024-09-21 by SayedaliSayedb