dallmann-consulting/OCPP.Core #74: can't connected to EVcharger

SayedaliSayedb, 2024-09-24:
How can I connect the charger device to this project?

[image]
[image]

[image]

SayedaliSayedb, 2024-09-24:
no this charger supported ocpp 1.6 and ocpp 2.0 and i can set model ocpp

dallmann-consulting, 2024-09-24:
The screenshots above don't say anything about OCPP. Maybe in "Communication settings"?

SayedaliSayedb, 2024-09-24:
in Communication settings we can set model OCPP (1.6 or 2 or other model)
we can set 

1-Remote Ip
2-Port Number
3-Mac Address

(in serial number setting) 
4-Server DominName
5-PileSerial Number

How We Can Set Root (eg 192.168.1.8/OCPP/model3)
How We can Set ID(model3) in this Device

JeremyMahieu, 2024-09-24:
I don't think remote IP means the public IP, maybe it's the IP of the OCPP server. But ye how are we supposed to figure out what the settings of this chinese charging point are? You didn't even show screenshots of all screens. Read the manual of your charging point, I'm sure it's well written.

SayedaliSayedb, 2024-09-24:
I have checked different parts
But apart from these two items, I did not find any other item on the page
Would it be possible to advise what the settings are like on other devices and what I should be looking for?

SayedaliSayedb, 2024-09-24:
How are the settings on other devices?

dallmann-consulting, 2024-09-24:
Please ask the maker of this charger. For other chargers you can find manuals in the internet.

closed 2024-09-25 by dallmann-consulting