dallmann-consulting/OCPP.Core #75: cant run on server

SayedaliSayedb, 2024-09-25:
Hello  Dear
I cant Run on Server
Is it possible to check where the problem is?

[image]

[image]

SayedaliSayedb, 2024-09-25:
I downloaded the [release](https://github.com/dallmann-consulting/OCPP.Core/releases). file and only changed the localhost settings

dallmann-consulting, 2024-09-25:
Please check if the necessary runtime (like IIS Hosting bundle?) is installed.

SayedaliSayedb, 2024-09-25:
it I was installed
Can you help me what's rong
Spell uid and pwd or other is rong ?

JeremyMahieu, 2024-09-25:
Start it in cmd first and show us the output.
```
cd your/install/directory/OCPP.Core.Server
dotnet OCPP.Core.Server.dll
```

SayedaliSayedb, 2024-09-25:
info: Microsoft.EntityFrameworkCore.Database.Command[20101]
      Executed DbCommand (31ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
      SELECT 1
info: Microsoft.EntityFrameworkCore.Database.Command[20101]
      Executed DbCommand (18ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
      SELECT OBJECT_ID(N'[__EFMigrationsHistory]');
info: Microsoft.EntityFrameworkCore.Database.Command[20101]
      Executed DbCommand (2ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
      SELECT 1
info: Microsoft.EntityFrameworkCore.Database.Command[20101]
      Executed DbCommand (1ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
      SELECT OBJECT_ID(N'[__EFMigrationsHistory]');
info: Microsoft.EntityFrameworkCore.Database.Command[20101]
      Executed DbCommand (3ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
      SELECT [MigrationId], [ProductVersion]
      FROM [__EFMigrationsHistory]
      ORDER BY [MigrationId];
info: Microsoft.EntityFrameworkCore.Migrations[20405]
      No migrations were applied. The database is already up to date.
fail: Microsoft.Extensions.Hosting.Internal.Host[11]
      Hosting failed to start
      System.InvalidOperationException: The non-HTTPS endpoint HttpsInlineCertFile includes HTTPS-only configuration for Certificate.
         at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.ConfigurationReader.ThrowIfContainsHttpsOnlyConfiguration(EndpointConfig endpoint)
         at Microsoft.AspNetCore.Server.Kestrel.KestrelConfigurationLoader.Reload()
         at Microsoft.AspNetCore.Server.Kestrel.KestrelConfigurationLoader.LoadInternal()
         at Microsoft.AspNetCore.Server.Kestrel.Core.KestrelServerImpl.BindAsync(CancellationToken cancellationToken)
         at Microsoft.AspNetCore.Server.Kestrel.Core.KestrelServerImpl.StartAsync[TContext](IHttpApplication`1 application, CancellationToken cancellationToken)
         at Microsoft.AspNetCore.Hosting.GenericWebHostService.StartAsync(CancellationToken cancellationToken)
         at Microsoft.Extensions.Hosting.Internal.Host.<StartAsync>b__15_1(IHostedService service, CancellationToken token)
         at Microsoft.Extensions.Hosting.Internal.Host.ForeachService[T](IEnumerable`1 services, CancellationToken token, Boolean concurrent, Boolean abortOnFirstException, List`1 exceptions, Func`3 operation)
Unhandled exception. System.InvalidOperationException: The non-HTTPS endpoint HttpsInlineCertFile includes HTTPS-only configuration for Certificate.
   at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.ConfigurationReader.ThrowIfContainsHttpsOnlyConfiguration(EndpointConfig endpoint)
   at Microsoft.AspNetCore.Server.Kestrel.KestrelConfigurationLoader.Reload()
   at Microsoft.AspNetCore.Server.Kestrel.KestrelConfigurationLoader.LoadInternal()
   at Microsoft.AspNetCore.Server.Kestrel.Core.KestrelServerImpl.BindAsync(CancellationToken cancellationToken)
   at Microsoft.AspNetCore.Server.Kestrel.Core.KestrelServerImpl.StartAsync[TContext](IHttpApplication`1 application, CancellationToken cancellationToken)
   at Microsoft.AspNetCore.Hosting.GenericWebHostService.StartAsync(CancellationToken cancellationToken)
   at Microsoft.Extensions.Hosting.Internal.Host.<StartAsync>b__15_1(IHostedService service, CancellationToken token)
   at Microsoft.Extensions.Hosting.Internal.Host.ForeachService[T](IEnumerable`1 services, CancellationToken token, Boolean concurrent, Boolean abortOnFirstException, List`1 exceptions, Func`3 operation)
   at Microsoft.Extensions.Hosting.Internal.Host.StartAsync(CancellationToken cancellationToken)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.Run(IHost host)
   at OCPP.Core.Server.Program.Main(String[] args) in C:\Users\ulrichd\source\repos\GitHub\OCPP.Core\OCPP.Core.Server\Program.cs:line 42

dallmann-consulting, 2024-09-25:
There is a problem with the SSL hosting. Try to remove the https endpoint from the appsettings:

[image]

And then open the browser with a normal "http" url (without "s").

For more details you can google the error message:
_Hosting failed to start System.InvalidOperationException: The non-HTTPS endpoint HttpsInlineCertFile includes HTTPS-only configuration for Certificate._

JeremyMahieu, 2024-09-25:
[image]
This should be https.

closed 2024-10-03 by dallmann-consulting