dallmann-consulting/OCPP.Core #80: Middleware Error

BurakOgutken, 2024-10-15:
First of all, great job on the OCPP backend software you've created. We are trying to integrate our station software with your backend, but intermittently, we are encountering the error I shared below on the middleware side. We haven't been able to understand the cause. Could you help us figure it out?

```
 Microsoft.AspNetCore.Server.Kestrel[13]
      Connection id "0HN7D91NRVPPK", Request id "0HN7D91NRVPPK:00000001": An unhandled exception was thrown by the application.
      System.Net.WebSockets.WebSocketException (0x80004005): The WebSocket is in an invalid state ('Aborted') for this operation. Valid states are: 'Open, CloseReceived, CloseSent'
         at System.Net.WebSockets.WebSocketValidate.ThrowIfInvalidState(WebSocketState currentState, Boolean isDisposed, WebSocketState[] validStates)
         at System.Net.WebSockets.ManagedWebSocket.CloseAsync(WebSocketCloseStatus closeStatus, String statusDescription, CancellationToken cancellationToken)
```

dallmann-consulting, 2024-10-15:
It doesn't seem related to OCPP.Core but to websockets in general. You can google it yourself...
The websocket specification names some scenarios how connections come to an aborted state:
https://datatracker.ietf.org/doc/html/rfc6455

dallmann-consulting, 2024-11-05:
Did you find a solution?