dallmann-consulting/OCPP.Core #83: 客户端意外断开连接后，服务器未针对心跳超时关闭websocket连接

FuYan, 2024-11-05:
客户端意外断开连接后，服务器未针对心跳超时关闭websocket连接，导致_chargePointStatusDict 还存储了状态数据，而且此时通过api发送websocket消息会出现异常
private static Dictionary<string, ChargePointStatus> _chargePointStatusDict = new Dictionary<string, ChargePointStatus>();

dallmann-consulting, 2024-11-05:
I'm not sure what the problem is. Your description is more than short.
The OCPP server does not manage the websocket connections. It relies on the ws connection status. If the connection closes the status in the dictionaray is removed. If a connection silently fails there might be a timeout interval until the connection officially gets closed. Of course all API calls will fail here as well.
But after both sides have noticed the failure they will close the dead connections and the charger can connect again. Thats the same for every tcp connection.

So where exactly is a problem in the OCPP server?

FuYan, 2024-11-06:
If the client unexpectedly disconnects and the dictionary is not removed, the management backend will display that the device is online when obtaining the status

dallmann-consulting, 2024-11-07:
I couldn't really reproduce the issue on one machine. The connection always got closed friendly.
Please check the latest changes if that helps.