dallmann-consulting/OCPP.Core #91: Update auf ocpp2.0.1

StBoll, 2025-06-14:
Seit dem neuesten Update der Keba x30 auf dir Firmware 1.9.0 wurde in der Software die OCPP Version 2.0.1 integriert, welches OCPP.Core scheinbar noch nicht unterstützt.

warn: OCPPMiddleware[0]
      OCPPMiddleware => No supported sub-protocol in ',ocpp2.0.1' from charge station '25464003'

Wird es in nächster Zeit ein Update geben, oder muss ich die Wallboxen wieder zurück auf OCPP 1.6 setzen?

dallmann-consulting, 2025-06-17:
Die Korrektur ist jetzt im Source bereits enthalten und ich werde morgen die Funktionen mit meiner Keba Wallbox testen. Mit dem Simulator funktioniert es, aber vielleicht sind noch weitere Abweichungen im echten Leben vorhanden.

closed 2025-07-12 by dallmann-consulting

dallmann-consulting, 2025-07-13:
Ja, ich war schon dabei, hatte dann aber noch einen kleinen Fehler in der Darstellung der Ladeleistung entdeckt. Den wollte ich noch kurz beseitigen. V1.5.0 steht jetzt bereit.