dallmann-consulting/OCPP.Core #95: idTag not showing in logs

netomx, 2025-07-04:
Hi.

While debbugging a charger, when the charger scans the rfid and send it to the server, it doesnt show in the log:

`info: Microsoft.EntityFrameworkCore.Database.Command[Microsoft.EntityFrameworkCore.Database.Command.CommandExecuted] @ 2025-07-03T23:42:32.6300441-06:00
      => SpanId:1e2863c16885bd2a, TraceId:3f51717b36a7d852673cc0162fc6695b, ParentId:0000000000000000 => ConnectionId:0HNDQJ263920L => RequestPath:/ocpp/1993 RequestId:0HNDQJ263920L:00000001
      Executed DbCommand (0ms) [Parameters=[@p0='?' (Size = 4), @p1='?' (DbType = Int32), @p2='?', @p3='?' (DbType = DateTime), @p4='?' (Size = 9), @p5='?'], CommandType='Text', CommandTimeout='30']
      INSERT INTO "MessageLog" ("ChargePointId", "ConnectorId", "ErrorCode", "LogTime", "Message", "Result")
      VALUES (@p0, @p1, @p2, @p3, @p4, @p5)
      RETURNING "LogId";
`

I cannot diagnose if the id of the tags are correct or not. 

Any ideas on how to show the id of the scanned tag?

Thanks

dallmann-consulting, 2025-07-05:
Yes, authentication details should probably be written to the message log. But currently they are not. And you currently can't see the message log in the UI. So you need technical access anyway.
The server can write all messages into files - see appsettings file. You will find the incoming OCPP message there incuding the RFID-ID.

dallmann-consulting, 2025-07-06:
I guess there is something wrong with your installation. Can you describe what you are doing exactly?
- Do have e real charger or using a simulator?
- Does the charger/simulator connect successfully?
- If the connection is successfull - does the server get heartbeat messages?
- Does the server write a log file? Can you delete the log file and make a fresh connect and then trigger the authentication (RFID)? And then please post this log here.

netomx, 2025-07-08:
Real charger.
Yes.
Yes.
Yes.

https://pastebin.com/T7HqEJmS

dallmann-consulting, 2025-07-09:
Sorry, but this is not the complete log. There is an important part missing at the beginning.

You should see an entry like this in the log:
```
dbug: Authorization[0] @ 2025-07-09T19:06:00.8153377+02:00
      Authorize => action='Authorize'; token='B4A63CDF'; chargePointId='Test1234'; connectorId='0'; transactionId=''; transactionStartToken=''
```
You see the the RFID number in the token field.

And in the dump directory there should be an authorization message:
`[2,"45JZrBZYdsjrCq8AgYYdfDvT7xcysTmF6aUu","Authorize",{"idTag":"B4A63CDF"}]'`