dallmann-consulting/OCPP.Core #97: beide Anschlüsse werden als belegt angezeigt, obwohl nur der 2. Anschluss belegt ist

StBoll, 2025-08-31:
[image]

[image]

[image]

dallmann-consulting, 2025-09-01:
Das passiert mit OCPP 2.0, richtig? Den Effekt hatte ich auch schon bemerkt. Die Ladetransaktion ist korrekt abgeschlossen (=> Anzeige der geladenen kWh anstelle der Ladezeit), aber der angezeigte Status ist verkehrt.

dallmann-consulting, 2025-09-29:
Ich kann das Verhalten leider nicht nachvollziehen. Weder mit Simulator, noch mit der echten KEBA-Box. 
Ich habe in meinen Nachrichten gesehen, dass die KEBA-Box den Status schon 1-2 Sekunden vor dem Ende der Transaktion setzt. Aber auch das führt dazu, dass der Status unmittelbar "grün" wird.
Wie genau beenden Sie die Transaktion? Vielleicht passiert dies nur in einem speziellen Szenario, in welchem die KEBA-Box kein Status-Update übermittelt?

StBoll, 2025-10-07:
Nachtrag:
Wird der Ladevorgang beendet, werden beide Anschlüsse auch wieder grün angezeigt. 
Es wirkt ein bisschen so, als würde nicht die Anschluss ID, sondern die Kennung für den Status genutzt. Mit 1.6 funktioniert alles fehlerfrei.